raspiblitz/raspiblitz #805: Set IP address

Lupus-Pluvia, 2019-09-30:
How can I set the IP address so that it does not change when the Internet connection is lost?
[image]

ZeusMinos, 2019-10-03:
If you ISP change your wan ip you may consider using DynDNS

Lupus-Pluvia, 2019-11-07:
Thanks, I'll give it a try

rootzoll, 2019-11-07:
DynDNS makes it easier to address your node in remote Wallets like ZAP, but it will still create a changing IP in your 1ML history. To have a stable IP/address to the rest of the lightnig network with the RaspiBlitz there are at the moment two ways:

1) TOR - switching to TOR will give you stable onion-address and tell that to the Lightning Network that will stay the same, even if your ISP gives you a new IP.

2) SSH-Reverse Tunnel - the RaspiBlitz has a experimental feature that you can use a server with a stable IP on the internet as address proxy and get the traffic from some specific ports forwarded to your RaspiBlitz at home. That ways your IP address to the LN stays stable even if your ISP changes your local IP. For details see: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-setup-port-forwarding-with-a-ssh-tunnel

rootzoll, 2019-11-07:
Please add further comments .. but I will close this issue for now.

closed 2019-11-07 by rootzoll