raspiblitz/raspiblitz #809: Touchscreen calibration

frennkie, 2019-10-04:
On my Raspberry Pi 4 I had to look up the calibration procedure again which we already discussed in #646. 

I would suggest to install the `xinput_calibrator` tool by default (`build_sdcard.sh`).

For the setup procedure we should either create a menu entry or a bash script. The procedure is simple but - of course - needs user interaction on the screen.

Start tool and let user calibrate:

```
sudo -u pi DISPLAY=:0.0 xinput_calibrator --output-filename /tmp/99-calibration.conf
```

Copy data to xorg config (**attention**: check for other things that touch this file - e.g. rotation)

```
sudo cp /tmp/99-calibration.conf /etc/X11/xorg.conf.d/99-calibration.conf
init 3
sleep 1
init 5
```

openoms, 2019-10-07:
Unfortunately the touch on my screen is all over the place and even when the pointer would move I couldn`t 
I have the usual one from the shopping list with the `XPT2046 Touch Controller`

Workflow before attempting the calibration: 
* Enable Touchscreen from the `SERVICE` menu. (otherwise error: `Unable to connect to X server` )
* Disable LCD rotate - otherwise my screen does not sense any touch (pointer does not move at all)
* Install command for the `xinput-calibrator`:
`sudo dpkg -i /home/admin/LCD-show/xinput-calibrator_0.7.5-1_armhf.deb`

* when running: 
`sudo -u pi DISPLAY=:0.0 xinput_calibrator --output-filename /tmp/99-calibration.conf`
The message suggests there is a `get_precalib.sh':
```
Calibrating standard Xorg driver "ADS7846 Touchscreen"
	current calibration values: min_x=0, max_x=65535 and min_y=0, max_y=65535
	If these values are estimated wrong, either supply it manually with the --precalib option, or run the 'get_precalib.sh' script to automatically get it (through HAL).
```
I could not find that on my RPi or in https://github.com/goodtft/LCD-show.
Did you need any further steps I should do to get the touchscreen working (like tusing the `ts_calibrate` described here: http://www.circuitbasics.com/raspberry-pi-touchscreen-calibration-screen-rotation/) ?

I will have another screen soon, will test that too.

frennkie, 2019-10-07:
I can't recall ever having used `get_precalib.sh`.

I'm not sure whether you need to Disable LCD rotate.. my config after calibration looks like this (take not of the SwapAxes):

```
Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "ADS7846 Touchscreen"
        Option  "Calibration"   "239 3911 3852 279"
        Option  "SwapAxes"      "1"
EndSection
```

openoms, 2019-10-07:
I will try to substitute your setting to the conf file, it might be that my touch settings are off by too much.
Do you have a shim fan on the GPIO under the screen (just out of curiosity)?

frennkie, 2019-10-07:
> Do you have a shim fan on the GPIO under the screen (just out of curiosity)?

Yes, I do.

openoms, 2019-10-10:
Good news, my new screen (with power pins and a fan under it) is working out of the box!

frennkie, 2019-10-19:
@rootzoll this somehow needs to be done when touchscreen is activated (or at least an option on the menu).

rootzoll, 2019-11-20:
All touchscreens I had were almost working good by default. I think so we dont need to force a calibration - but it should be an option in the hardware sub menu for the future.

rootzoll, 2020-01-26:
OK I added touchscreen calibration as an option to the main menu when touchscreen is activated.
Tested and works fine. Closing Issue.

jfrader, 2021-11-01:
Im having this same issue and I don't think its a broken screen because when calibrating with the menu option  clicking works in every corner, but after config is saved, its working only in one side and not calibrated at all.

rootzoll, 2021-11-11:
We will recheck the touchscreen calibration for v1.8 because there might be changes coming with v1.8 that affect touchscreen anyway.

JoKohono, 2022-01-05:
My touch screen does not respond to any touch (finger/stylus). I have the waveshare 3.5“ (Rev 2.0) and made the modifications to the overlay file (dtoverlay=waveshare35b-v2) which fixed the inverted-color issue. 
But I‘m left with no-touch functionality. 
(Screen looks good with all the Info/Node/Invoice/Off-buttons). 
Any idea?

rootzoll, 2022-01-08:
I have tested the touchscreen calibration for the upcomming v1.7.2 release - it basically worked.
To be retested to next Release Candidate release.

rootzoll, 2022-02-17:
Calibration tested OK again on the v1.7.2RC2 ... will now push this issue up to v1.8.1 milestone where the touchscreen will get more focus again,