raspiblitz/raspiblitz #810: LND Hub improvement

yannick10800, 2019-10-06:
Hello
Could you add the feature of LNDhub ? (such as https://github.com/dangeross/guides/blob/master/raspibolt/README.md)

To allow the Node to be a LND hub?
This allows user to be hub for different people, which reduces the cost of opening channels for end users (but is more centralized)
Thanks

jspeigner, 2019-11-20:
+1 for this feature

openoms, 2019-11-21:
Relaying from TG chat: 
this guide most likely works on the Blitz too:
https://github.com/dangeross/guides/blob/master/raspibolt/raspibolt_6B_lndhub.md
Someone could try.
The port 3000 is used by RTL so either only run one of the those services at a time or change the config of either.

We should also consider privacy implications when using LNDHub, but as an optional feature it could be good.

nyxnor, 2021-06-19:
This is a duplicate of the issue openoms mentioned above. pls close this one.

closed 2021-07-01 by rootzoll