raspiblitz/raspiblitz #815: [lncli] unable to read macaroon path

lutzk87, 2019-10-07:
Hey, got the same issue on v1.1 and the new v1.3 version.

Im trying to setup a Litecoin Full Node. Litecoin Core v0.16.3 runs just fine but i cant connect to any lightning node.

Got only this issue [lncli] unable to read macaroon path (check the network setting!): open /home/admin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory

Hope someone can help me.

Hardware: Raspberry Pi 3B

frennkie, 2019-10-07:
If you use litecoin than the path should be litecoin and not bitcoin... I assume you have to pass `chain=litecoin` to `lncli`.

lutzk87, 2019-10-07:
i know that the path should look like this:

 /home/admin/.lnd/data/chain/litecoin/mainnet/admin.macaroon

iam running on the litecoin chain:

"identity_pubkey": "0294a28242d2cd106202a74d35836e9c7790863dd82d23194e73133c6951c532a3",
    "num_pending_channels": 0,
    "num_active_channels": 0,
    "num_peers": 0,
    "block_height": 1715442,
    "block_hash": "961cfed45207ac0e28cd81c073bd0c1d0af5e5b09b1c08bc97e61fcd032da1a7",
    "synced_to_chain": true,
    "testnet": false,
    "chains": [
        "litecoin"
    ],
    "uris": [
        "0294a28242d2cd106202a74d35836e9c7790863dd82d23194e73133c6951c532a3@185.141.64.106:9735"
    ],
    "best_header_timestamp": "1570470945",
    "version": "0.5.2-beta commit=v0.5.2-beta",
    "num_inactive_channels": 0

frennkie, 2019-10-07:
Looks fine to me... What is your problem..? Or maybe when does it occur (steps to reproduce... )?

lutzk87, 2019-10-07:
i cant run lncli getinfo same issue 

admin@RaspberryPi3BLTCFullNode:~ $ lncli getinfo
[lncli] unable to read macaroon path (check the network setting!): open /home/admin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory

i cant connect to any lightning node, if i want to the same error occurs.

admin@RaspberryPi3BLTCFullNode:~ $ lncli connect 02d90162819b0f5b8b0f8ae755eb4c0e0fea3cbf6ed7a0bc19226fa01aba7323f3@84.149.141.140:9735
[lncli] unable to read macaroon path (check the network setting!): open /home/admin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory

u can change the macaroon path with the command lncli -- --macaroonpath=/home/admin/.lnd/data/chain/litecoin/mainnet/admin.macaroon is this right?

when i do this nothing happens. i just see the help menu again of lncli.

frennkie, 2019-10-07:
Try `lncli chain=litecoin getinfo`...?!

If that works then it would be `lncli chain=litecoin connect... `

lutzk87, 2019-10-07:
admin@RaspberryPi3BLTCFullNode:~ $ lncli chain=litecoin getinfo
No help topic for 'chain=litecoin'

frennkie, 2019-10-07:
Please use try to use the help from to time...

My example is missing the  `--`.

So `lncli - - chain=litecoin getinfo`

lutzk87, 2019-10-07:
u are right mate. that works. i remember some other version (v0.99) where this issue doesnt appear.

u helped me alot. thanks very much!

frennkie, 2019-10-07:
@lutzk87 please upgrade your lnd to at least version 0.7.1! There was a security issue that was fixed recently.

closed 2019-10-07 by frennkie

lutzk87, 2019-10-07:
will do that, i just testet the version of raspiblitz v1.1 because the same issue on v1.3 occurs. TY!