raspiblitz/raspiblitz #816: Closing just one channel

PBJellytime, 2019-10-07:
I've got a dead channel but can't seen to close just the one channel using lncli and no options on the main screen except close all channels

frennkie, 2019-10-07:
What happens when you try to close the channel using `lncli`? Error message?

PBJellytime, 2019-10-07:
admin@EmJ:~ $ lncli closechannel 655711351584391169
[lncli] rpc error: code = Unknown desc = channel not found

frennkie, 2019-10-07:
Can you run `lncli listchannels`. 

Then use the **channel_point** value (replace the ':' with a space).

PBJellytime, 2019-10-07:
That worked, except I need to force close :p 

admin@EmJ:~ $ lncli closechannel 1ccbbb150fb2985abe49fe133debc96580cd771dbd39c8dba0184f475b835493 1
[lncli] rpc error: code = Unknown desc = unable to gracefully close channel while peer is offline (try force closing it instead): channel link not found

PBJellytime, 2019-10-07:
Thanks for the help guys o7

closed 2019-10-07 by PBJellytime