raspiblitz/raspiblitz #817: run raspiblitz in bitcoin Pruned mode

woeisme, 2019-10-08:
I'd like to switch my raspiblitz install to a pruned node but I'm not sure the best method or if it will mess up the associated LND wallet. 

Is it just as simple as modifying bitcoin.conf to add the prune= parameter and rebooting?

openoms, 2019-10-09:
Yes, LND should be able to work with pruned mode. Electrs (if it is installed) does not.
I think one of the reasons of run a dedicated hardware is that there is no need for pruning so not recommending.

fluidvoice, 2019-10-09:
> I'd like to switch my raspiblitz install to a pruned node but I'm not sure the best method or if it will mess up the associated LND wallet.
> 
> Is it just as simple as modifying bitcoin.conf to add the prune= parameter and rebooting?

have a look at this issue thread: https://github.com/rootzoll/raspiblitz/issues/268

woeisme, 2019-10-25:
> 
> 
> > I'd like to switch my raspiblitz install to a pruned node but I'm not sure the best method or if it will mess up the associated LND wallet.
> > Is it just as simple as modifying bitcoin.conf to add the prune= parameter and rebooting?
> 
> have a look at this issue thread: #268

I've looked at thread @fluidvoice but I'm still uncertain if it's safe to prune from the content, as I plan on keeping this node running for a while. Do I then need to prune but keep 1 years worth of data?

Are there any other downsides I'm not realizing with a pruned node? 
Can I do this on a active full (tor based) node with open lnd channels, merely by adding the 'prune' option to bitcoin.conf and rebooting the RPI4 ?

nyxnor, 2021-04-18:
@woeisme Have you tried it?
If it worked, please let us now and close this.