raspiblitz/raspiblitz #818: Security: How to verify the EEPROM of the RPI4?

openoms, 2019-10-09:
As Damian brought to our attention the RPi4 has a writable 512 KB EEPROM which can be tampered with: https://twitter.com/meeDamian/status/1179052293053792256
From the docs (https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md):
```
There is no software write protection for the boot EEPROM but there will be a mechanism in Raspbian to skip any future updates to the EEPROM.
```
How could we verify what is there?
also: 
```
Note that if a bootcode.bin is present in the boot partition of the SD card in a Pi 4, it is ignored.
```
The Raspbian images contain a bootcode.bin so the EEPROM should be ignored when it is present. I don`t know what governs the check for the bootcode.bin, so I for now I presume that malicious code in the EEPROM can override the SD.

Discussed in the RPi forum too: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=253843

openoms, 2019-10-10:
There is a recovery image provided with which the EEPROM can be updated if in doubtl: https://www.raspberrypi.org/downloads/
Unfortunately I could not yet find a signature to verify it`s content let alone verifying the EEPROM without flashing.

openoms, 2019-10-10:
There is a recovery image provided with which the EEPROM can be updated if in doubt: https://www.raspberrypi.org/downloads/
Unfortunately I could not yet find a signature to verify it`s content let alone verifying the EEPROM without flashing.

openoms, 2019-10-26:
Now there is a sha256 hash provided for the RPi4 bootloader image:

[image]

Will need to either reflash the boards with the verified image or find the way to clone it from the bootloader and check the hash.

xanoni, 2021-08-02:
@openoms any updates on this? Also, is there anything that can be done to protect one's machine from manipulation once it's running in production? (Understand you are just concerned about supply-chain attacks here.)