raspiblitz/raspiblitz #819: Node name and it's local usage

Conimous, 2019-10-09:
Hi,

When setting up the Raspiblitz it prompts for the naming of the LN node right? Y is this tight into the naming of the local machine? Why am asking? Well if you run ur Node on TOR but ur local machine blatantly blasts out its name to ur router, ISPs are able to read those out and figure out which nodes run where, based on the local name of the Raspi.

Why not set it to " Unknown device " or " Raspberry " on the local side of the machine, especially when configured to run on TOR. 

Ty in advance for the responses.

openoms, 2019-10-09:
Agreeing here, we might want to keep the default hostname `raspberry` or at least change it back if Tor is activated.
The MAC adress is known to the router, so the device type is exposed to the ISP (through a proprietary router) but at least the type of usage should not be advertised.

Relevant code: https://github.com/rootzoll/raspiblitz/blob/master/home.admin/config.scripts/lnd.setname.sh

rootzoll, 2019-12-10:
I removed setting the local networkname by default on fresh setups now. For old setups it will keep the old local network name to not break possible forwarding rules on the routers if configured by hostname.

rootzoll, 2020-01-23:
OK. Works on my test setups of fresh devices. Closing issue.

closed 2020-01-23 by rootzoll