raspiblitz/raspiblitz #820: Raspiblitz 1.3 runs with errors even after several reboots

nanguapie, 2019-10-09:
I have been running a well working Raspiblitz 1.1 since March 2019 on a Raspberry Pi 3+. 

This has been a great experience and the node has been very stable and both routed, sent and received many transactions. 

As soon as I upgraded to Raspiblitz 1.3 though, the node has not been able to run for more than half an hour or so at a time. When SSL-ing in again after an hour or so, I get the warning that the node is "running with errors".  I first thought it was due to overheating, so I removed the screen from the node setup and the running temperature now stays around 60 C which I believe should be fine. 

I also ran the hardware test and no issues were discovered through that. 

I attach the debug log here. Please let me know if you have any thoughts on what I can do to get the node up and running again. 

I have a number of channels open , but of course 0 of them are active at the moment. And it seems that my node does not have time - or doesnt succeed for any other reason - to re-activate them when I re-boot the node. The nr of active channels stays at 0 until the node fails again. 

Many thanks for support! 

Here is the full debug log: 
https://termbin.com/ttlc

openoms, 2019-10-10:
Hi, the problem seems to be with lnd. 
Sometimes when there is a big version jump the migration process can have issues, although I did not see this error before:
```
2019-10-09 19:51:35.252 [ERR] LTND: Unable to start server: unable to fetch node: 02a4ff88f99ec4491b108f5d19d1bf2ea855342f1e646115187218c4e9b83bcefa, unable to find node
```

You could try to reflash your SDcard to trigger the restore process again which has a chance to solve this problem.

You can make a full backup of your lnd directory first by running:
`/home/admin/config.scripts/lnd.rescue.sh backup`

Let us know here how it goes. If the same error persists it might be worth to ask in the LND Developer Slack: https://app.slack.com/client/T6AK88MGV/ or https://github.com/lightningnetwork/lnd/issues before progressing to any of the recovery processes: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-can-i-recover-my-coins-from-a-failing-raspiblitz .

nanguapie, 2019-10-18:
Thank you for taking the time to look into this and answering! 

I tried your advice to reflash the SDcard from scratch - I even downloaded the Raspiblitz v 1.3 image again. It worked just as well as last time flashing the SD card and rebooting the system, it synced fully both bitcoin and lightning .  But then when running the system, still none of my open channels are connected.  And after an hour or so, when SSL:ing in again the wallet was locked , which to me means that the wallet has been closed due to some error. Since this behaviour I never saw with the version 1.1, which always worked perfectly. Now i tried again, and again the wallet was locked. 

The node is running now, but still failing to liven up any of the channels.  this is my debug log as of now:
https://termbin.com/zxac 

Is there any reason not to erase the entire harddrive and then use the LND seed word to set up the node again from scratch? I'm just thinking that that might be easier than trying to find the issue on the Developer Slack and then fixing it.. Or would resetting the entire setup lead to loss of funds or other properties of the node that i should keep?  Not sure if I should actually close all channels and remove all funds before doing that. The annoying part would be to pay all the non-cooperative closing fees, which im not sure if i can actually control myself to only set 1 sat per byte. 

thank you for further advice! I am no expert as you notice, but I find the raspberry / LND testing out quite fun from time to time, and the Raspiblitz is a cool packaged way of doing it during the times when it works! (most of the time so far) :)

nanguapie, 2019-10-26:
I just let it run with error until now due to no time to tinker, and the current debug log can now be found here. 

Please let me know if you have any advice as to what my next step should be. I still have some sats both on the onchain wallet and in the open channels of the node. Many thanks! 

https://termbin.com/jhve

openoms, 2019-10-27:
@nanguapie your lnd is still stuck below the block 600000 with the same error.
Not sure why this is, my best guess is database corruption.

Restoring your wallet on a new setup is probably your best bet. You will need: 
* the seed - for your onchain funds - and 
* Static Channel Backup (SCB - channel.backup) - for the offchain (channel) funds
your channels with online peers will be force closed and returned onchain.

Read this about the channel backup: https://github.com/rootzoll/raspiblitz/blob/master/README.md#backup-for-on-chain---channel-funds

Run:
`./98repairMenu.sh`

Make a backup of your LND directory.
Extract the rescue file and find you channel.backup at the path: `/mnt/hdd/lnd/data/chain/bitcoin/mainnet/`
When you have:
* the rescue file
* your 24 words seed and 
* channel.backup
 it is safe to reset your LND.

Run again:
`./98repairMenu.sh`
and reset your LND.
From there you can either:
* attempt to recover from the rescue file (unlikely to work if the problem is database corruption) or if it fails:
* recover from seed + SCB (will force close the channels).

nanguapie, 2019-11-03:
@openoms , thank you very much for taking the time and guiding me with restoring the node! The method you described worked fine, I recovered from seed + SCB and thus force closed the channels. And now the node is stable, finally correctly upgraded to Raspiblitz version 1.3! 

I have a couple of reflections though and one final question regarding the channel management - there may be lessons here to learn for next time: 

- I had 15 channels open when the failed upgrade happened, and I didn't work on fixing it for a few weeks. This may have led to changes happening on the channels without my node's knowledge and thus I may have been punished by losing all funds in some channels. 

- As of now, I have got approximately half of my funds that were previously in channels now on the on-chain wallet due to what you call above the SCB "force closing" of the channels.  I don't believe actually what happened was a true "force close", since now that I look in the RTL (Ride the Lightning) interface, I can see that there are only 6 channels listed which are all under "Pending" -> "Waiting Close Channels(6)".  My point is that they are not under "Pending Force Closing Channels(0)" 

- If these closes eventually become successful (should they be successful after 144 blocks, or is the standard setting more?) then I might get more of the channel funds back, which I of course hope for. If not, I guess that they were donated to the friendly channel managers on the other side...

- Also, this whole manoeuvre of course incurred 15 on-chain fees for closing the channels. Was there any way in which I could have affected the fee level of those closings? When you close a channel manually you can choose 1 satoshi per byte and just wait a few hours/days/weeks until the mempool is more or less empty, which would be fine in this case.  But now I have no idea what I paid for the channel closings.  This I believe is an improvement area for LND to make on-chain fees more transparent and easy to manage in all different scenarios.  Fees should be a highly prioritized area and if fees start to rise again, there is a risk for a lot of user annoyment if by "automation" users suddenly pay really hurtful fee levels for tasks such as closing channels which is not time critical at all.  

Please let me know if you think that is worth reporting as issue to LND and I can do it in that case.

Thanks again for all support!

closed 2019-11-10 by nanguapie

openoms, 2019-11-25:
You can.see what you paid for the closing transactions by looking at the UTXO you received back in.a block explorer. 
You would need to look up the tx fee of the incoming transactions to the addresses where your coins are.

nanguapie, 2019-11-25:
Cool, thank you.  That is of course interesting to know what was paid for the closing of the channels, I will have a look. 

My question was more regarding the future though, if it's possible to actually choose when force closing channels? I know that it's possible in the command line interface when closing cooperatively, but perhaps not for the force closes.

nanguapie, 2019-11-25:
Just FYI @openoms ,  I see on the transactions that I paid between 18 and 20 sats/vByte for the force close transactions.  Clearly more than I would have for this type of non-urgent transaction, but not at all brutal. It would be great if it was possible to manually change that when force closing.  My restoration of the channel backup file seem however to have pushed the channel partner to force close from their side (in my RTL it says "REMOTE_FORCE_CLOSE" , so I guess I couldn't have affected this fee level anyway, although the channel partner possibly could have.

openoms, 2019-11-25:
These fees are automatically set by LND according to the current state of the mempool and AFAIK cannot be manually set. This is to make sure that there is no race condition to have an old state confirmed earlier by the other peer and make possible to cheat. This is why the commit fee and channel balances are seemingly changing when the mempool is filled.

nanguapie, 2019-11-25:
Aha, very good to know, thank you @openoms.