raspiblitz/raspiblitz #821: Can't ssh into raspiblitz after closing my terminal window.

mattcruzz, 2019-10-11:
I left my raspiblitz v1.3 syncing via torrent and then closed the terminal window on my mac. 

When I returned a few hours later and tried to ssh in I got this:
Connection reset by 192.168.3.2 port 22

On the raspiblitz screen, it flashes this across the whole screen:
[18600.378472] EXT4-fs error (device sda1): ext4_find_entry: 1455: inode #2: comm sshd: reading directory 1block 0

Did an error occur or do I just have to wait until its fully sync'd before I can get back in?

openoms, 2019-10-12:
Aew you using an RPI3 or 4? What power supply (what is the Amper output)?
This looks very much like a HDD error, possibly:
* the most common is a problem is with the power supply, need > 3A -> use a good one from the shopping list or a Y-cable with external power or powered USB hub
* faulty HDD - need to change an start fresh

You can asses the disk state connecting to  a Linux computer (usb booted Ubuntu fro example) or format it and run a surface check on you Mac.

mattcruzz, 2019-10-12:
I'm using a RPI4 and the suggested Official Raspberry Pi 4 Power Supply (USB-C).

I went ahead and plugged the disk into my Mac, erased its contents, and ran "First Aid" on it. It didn't report any errors. I'm going to reflash the SD card with v1.3 and try again.

mattcruzz, 2019-10-12:
Process:

1. I passed the hardware setup.
2. Proceeded to the Bitcoin node setup.
3. Entered name for the pi and passwords A & B.
4. Went to format the HDD:

*** Formatting the HDD ***
WARNING ALL DATA ON HDD WILL GET DELETED - CAN TAKE SOME TIME
Wait until you get a OK or FAIL
mke2fs 1.44.5 (15-Dec-2018)
/dev/sda1 contains a hfsplus file system labelled 'Untitled'
Creating filesystem with 244190208 4k blocks and 61054976 inodes
Filesystem UUID: f139afe9-9550-4af9-bea5-4acbda8cfccf
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736, 1605632, 2654208, 
	4096000, 7962624, 11239424, 20480000, 23887872, 71663616, 78675968, 
	102400000, 214990848

Allocating group tables: done                            
Writing inode tables: done                            
Creating journal (262144 blocks): done
Writing superblocks and filesystem accounting information: done     

format ext4 done - wait 6 secs
FAIL - was not able to format the HDD to ext4 with the name 'BLOCKCHAIN'

And now I'm not sure how to proceed.

mattcruzz, 2019-10-13:
After looking at #545 and seeing their voltage was ~1.2V and noticing mine is less than this, I need a better PSU.

> First off the power supply is weak. I recommend you buy a new one. 1.2V is def not enough and the test shows your RPi CPU is throttling (slowed down) significantly in an attempt to compensate for the low voltage.

closed 2019-10-13 by mattcruzz