raspiblitz/raspiblitz #823: No activity on node

PBJellytime, 2019-10-15:
I've had my node up for 3 weeks now, I have eleven channels open.  Two incoming and Nine out but checking my forwarding history nothing appears to have been routed through my node, nor have I had any connections made to my node, the two incoming are ones I requested to get inbound liquidity.  Its all green albeit I had to use TOR to get it to work

https://1ml.com/node/0275848f93ba643dc7f2ef4594463200de7a45329660b647040da4c5ef75a3e16f

Is there anything else I should be doing to kick start something happening to it?

nanguapie, 2019-11-04:
I've been running a node with 15 channels , approx half in and half out, for half a year and only routed 5-10 payments. A normal little "fringe" node like mine and yours just isn't that useful for routing as far as I can tell.  It's fun though once you do get some routes, so just make sure that you have low fees (both fixed and variable fees) set for your node, since most people do this for fun and not earning sats I think that fee levels are lower than they would've been without altruism..

rootzoll, 2019-11-16:
One idea of the RaspiBlitz is to be a node for friends and familiy .. so why not provide some inbound caopacity channels and good connectivity to the network for your friends and family then routing thru you. If you aim t play with the big routing nodes ... check the tutorials by Alex Bosworth on how to manage a Routing Node.

closed 2019-11-16 by rootzoll