raspiblitz/raspiblitz #824: Upgrade LND

BTCxZelko, 2019-10-15:
With Upgrade 0.8.0 out. How can we go about upgrading?

lutzk87, 2019-10-16:
Upgrade guide from openoms. worked for me!

> https://twitter.com/openoms/status/1184201843574034433?s=20

PBJellytime, 2019-10-16:
Upgrade worked for me, however when I try and view the channels now I get the below error message 

[lncli] credentials: failed to append certificates
[lncli] credentials: failed to append certificates
[lncli] credentials: failed to append certificates
[lncli] credentials: failed to append certificates
[lncli] credentials: failed to append certificates
[lncli] credentials: failed to append certificates
[lncli] credentials: failed to append certificates

mainnet                            Commit ------- Balance --------- --- Fee ----
Alias or Pubkey           Capacity   Fee        Local       Remote   Base PerMil
-------------------- ------------- ------ ------------ ------------ ----- ------
-------------------- ------------- ------ ------------ ------------ ----- ------
[lncli] credentials: failed to append certificates
Totals          0 ch             0      0            0            0 Day:
[lncli] credentials: failed to append certificates
                                                                    Week:
[lncli] credentials: failed to append certificates
                                                                    Month:

openoms, 2019-10-26:
@PBJellytime 
not sure what this error is, can you give more context?
Is it only the `lnchannels` command failing?
Is your node functional otherwise?

rootzoll, 2019-11-16:
@openoms can you confirm that the lnchannels is still working after the upgrade to lnd 0.8?

openoms, 2019-11-16:
> @openoms can you confirm that the lnchannels is still working after the upgrade to lnd 0.8?

Yes, `lnchannels` work perfectly on all my updated nodes.

openoms, 2019-11-24:
The new Bitcoin Core v0.19.01 is out https://twitter.com/bitcoincoreorg/status/1198559459401965568

LND v0.8.1 is an update to make it compatible with the above so we can update both for the next release.

frennkie, 2019-11-24:
Appears to working fine. Just on "gotacha": LND v0.8.1-beta identifies as v0.8.0-beta

```
$ lnd --version
lnd version 0.8.0-beta commit=v0.8.1-beta
$ sha256sum /usr/local/bin/lnd
654f222432dc3bae806032a4939d97eefc29e05d2ff380a5b9586ba6897dbc85  /usr/local/bin/lnd
```
They forgot to increase the version counter.

I updated the sd_build.sh script with: ff18754

openoms, 2019-11-24:
@frennkie did you do a manual SDcard build to test?

frennkie, 2019-11-24:
No, but will do so now.

openoms, 2019-11-24:
The update script works as well with LND v0.8.1-beta but the status screen continues to display `LND v0.8.0-beta`
https://github.com/openoms/raspiblitz-extras#lnd-update-to-v081-beta

openoms, 2019-11-27:
The latest LND is in the master branch now. Update scripts functional. Closing.

rootzoll, 2019-12-10:
LND 0.8.2 is already in RC2 ... expected to be final for v1.4 release. So I reopen this issue to check again closer to v1.4 release: https://github.com/LightningNetwork/lnd/releases

openoms, 2019-12-12:
Running LND v0.8.2-beta on two nodes. There was no issue with the update. https://twitter.com/openoms/status/1205250222164713472

rootzoll, 2019-12-19:
Tests look good with 0.8.2 ... closing issue.

openoms, 2020-01-13:
Upgrade to 0.9.0-beta-rc1 went ok; https://twitter.com/openoms/status/1216428426724356098

Update script: https://github.com/openoms/lightning-node-management/tree/master/lnd.updates#update-lnd-to-v090-beta-rc1

rootzoll, 2020-01-14:
Yeah looks like we need some extra days anyway to get rid of the bugs of v1.4 ... so why not wait for the final LND v.0.9 release.

rootzoll, 2020-01-22:
Final LND 0.9 release seems right around the corner.

rootzoll, 2020-01-22:
LND 0.9.0 is out .. adding it to the build script :D
https://github.com/lightningnetwork/lnd/releases/tag/v0.9.0-beta

openoms, 2020-01-22:
> LND 0.9.0 is out .. adding it to the build script :D
> https://github.com/lightningnetwork/lnd/releases/tag/v0.9.0-beta

And running it already, updated smoothly: https://github.com/openoms/lightning-node-management/tree/master/lnd.updates#update-lnd-to-v090-beta

openoms, 2020-01-22:
@rootzoll watch the signing keys it is @roasbeef who signed the release despite the Release notes pointing to Joost`s PGP: https://twitter.com/openoms/status/1220041314743595008

rootzoll, 2020-01-22:
OK - tests also run good on my machine. v1.4 will be released with LND v0.9.0-beta
closing issue

closed 2020-01-22 by rootzoll