raspiblitz/raspiblitz #829: Torrent blockchain outdated

m8tion-cell, 2019-10-20:
I am building a new raspiblitz (1.3) in order to recover an old broken one (1.0), using the tar.gz backup file. So far so good with the process but the blockchain downloaded using the torrent utility + updates is 4 months old (june 2019). It went from 92.28 to 92.38% blockchain sync in 2 hours with a very good bandwith, so that would be around ... 180 hours to sync. Is it planed to update the torrented blockchain ?

xsoft, 2019-10-24:
It took me like 3days from "last torrent state" to "100%" (on 500Mb line).

Well, this site also died at 05/2019 - https://getbitcoinblockchain.com/ (Torrent)

rootzoll, 2019-11-16:
I started also now seeding from my home computer again. Yeah getting the blockchain by torrent is almost a deprecated feature now that the RaspberryPi4 can selfvalidate and download the blockchain from the bitcoin network. There might be an update of the update-torrent file to keep old RaspberryPi3 in the game - but even for those I would recommend downloading/validating the blockchain yourself with your laptop and then copying it to the RaspberryPi -see details: https://github.com/rootzoll/raspiblitz#2-copying-from-another-computer

closed 2019-11-16 by rootzoll