raspiblitz/raspiblitz #833: LTC auto-unlock - not working (brick)

xsoft, 2019-10-24:
Hi, 
another feature/bug report.
https://github.com/rootzoll/raspiblitz#feature-overview

At 1.3, full synced, online. After using SERVICE > [6] LND Auto-Unlock, entering (C) password and autoreboot, the Pi will NOT boot to menu. It will end again at "Enter LND (C) password". Whatever password is entered (good/wrong), it will not check it, and just stuck at this state forever.

Basically by enabling this feature brick entire system.

rootzoll, 2019-11-16:
Added to v1.4 for testing and fix ...

rootzoll, 2019-12-19:
Tested with LTC and the new python3 migration. Auto-Unlock is working. New rework will be part of first v1.4 Release Candicate .. please retest.

closed 2019-12-19 by rootzoll

theotherside, 2019-12-26:
I was updating from v1.3 I had auto unlock for LND set. During new install v1.4. When the box came up where I had to reenter my password c. I accidentally hit the enter button and was not able to go back. The install went on to a point then went bad. I re wrote the sd card from image and restarted the upgrade everything was fine(did not accidentally miss entering password c). Maybe add a back button so if you miss entering you can try again? Or maybe I missed it.

rootzoll, 2020-01-06:
@theotherside OK this will be fixed with v1.4 RC2 thx