raspiblitz/raspiblitz #834: LND Service Running with Errors

PBJellytime, 2019-10-25:
Woke up this morning to a message saying LND had failed, rebooted and now showing this error message

****************************************
* The LND service is running with error.
****************************************
If you just started some config/setup, this might be OK.

More Error Detail:
unexpected fault address 0xe1530002 fatal error: fault [signal SIGSEGV: segmentation violation code=0x1 addr=0xe1530002 pc=0x3a7c20] goroutine 1 [running, locked to thread]: runtime.throw(0x9a2f69, 0x5) runtime/panic.go:774 +0x5c fp=0x2ca6e60 sp=0x2ca6e4c pc=0x40368 runtime.sigpanic() runtime/signal_unix.go:401 +0x310 fp=0x2ca6e78 sp=0x2ca6e60 pc=0x56718 regexp/syntax.init() regexp/syntax/perl_groups.go:115 +0x784 fp=0x2ca6f2c sp=0x2ca6e7c pc=0x3a7c20 runtime.doInit(0x11d3058) runtime/proc.go:5222 +0x88 fp=0x2ca6f40 sp=0x2ca6f2c pc=0x4e6a8 runtime.doInit(0x11d4ab8) runtime/proc.go:5217 +0x4c fp=0x2ca6f54 sp=0x2ca6f40 pc=0x4e66c runtime.doInit(0x11d74b0) runtime/proc.go:5217 +0x4c fp=0x2ca6f68 sp=0x2ca6f54 pc=0x4e66c runtime.doInit(0x11da060) runtime/proc.go:5217 +0x4c fp=0x2ca6f7c sp=0x2ca6f68 pc=0x4e66c runtime.doInit(0x11d63e0) runtime/proc.go:5217 +0x4c fp=0x2ca6f90 sp=0x2ca6f7c pc=0x4e66c runtime.doInit(0x11d97c8) runtime/proc.go:5217 +0x4c fp=0x2ca6fa4 sp=0x2ca6f90 pc=0x4e66c runtime.main() runtime/proc.go:190 +0x1bc fp=0x2ca6fe4 sp=0x2ca6fa4 pc=0x4229c runtime.goexit() runtime/asm_arm.s:868 +0x4 fp=0x2ca6fe4 sp=0x2ca6fe4 pc=0x6f4a8

PBJellytime, 2019-10-25:
I think it might be power supply issues so I've ordered a 5v 3amp one unless anyone else has any ideas?

rootzoll, 2019-11-16:
Having such a panic error is always hard to say whats going on ... updaing Power Supply is a good idea and I would update LND as much as possible ... if you dongt have 0.8 running yet, see the update script by openoms: https://github.com/openoms/lightning-node-management/blob/master/lnd.updates/README.md

closed 2019-11-16 by rootzoll