raspiblitz/raspiblitz #839: Borked RPI4 is possible to recover using spare RPI 2B

woeisme, 2019-10-27:
Shutdown my functioning RaspiBlitz,  RPI4 unit yesterday and for some strange reason it decided never to boot again after that. 

I'm worried about losing funds that I have in channels until I can move funds off could I migrate the sdcard and external harddrive to a RPI2 I have lying around just to get the system stable and booted once again?

Is there anything during setup that would prevent the RPI2 from using the latest project image?

Any help appreciated. This is now the 2nd RPI4 that has stopped working on me, frustrating.

woeisme, 2019-10-28:
https://termbin.com/j6bp

faket0shi, 2019-10-28:
powerFAIL=13
powerWARN=15
I think it might be power supply issues

RolfHodl, 2019-10-29:
RPI2 won't work coused by missing instructions sets of the cpu.

closed 2019-11-16 by rootzoll