raspiblitz/raspiblitz #846: Missing instructions for getting RPC password

guilsa, 2019-11-02:
This section in README is missing information on how to get the RPC password. I had to exit to shell on user admin and do cd ./bitcoin && less bitcoin.conf to get it. 

> If you like to do this all from a web browser with a dashboard UI instead from an SSH terminal, go to SERVICES, activate the RTL Webinterface and after the reboot open in your web browser: http://[LOCAL-IP-OF-YOU-NODE]:3000 (PASSWORD B is your RPC password).

rootzoll, 2019-11-07:
During Setup you should write down 4 Passwords: A, B, C & D ... like the README says later on - Password B is your RPC password. Are you missing the information how to recover the Password B if you have not written it down in the beginning?

nyxnor, 2021-04-18:
pls close this.

guilsa, 2021-04-18:
Apologies that I never responded. Lost context by now, closing.

closed 2021-04-18 by guilsa