raspiblitz/raspiblitz #849: Can't SSH to raspiblitz anymore. Can't find it on local network.

adolfosrs, 2019-11-04:
My node was working fine for something like 3 months but now I'm not being being to SSH to it. 
I'm not being able to see the raspberry connected to my local network as well and when restarting the node I see that my HD gets turned after a couple of minutes probably because of it gets iddle.

I don't have the rasp display so can't see any output from it.

I guess it will probably show up again if I reflash a new SD card image but don't want to do that since I would have some work to backup my channels.

Any guesses or suggestions?

rootzoll, 2019-11-07:
Reflashing the sd card is a valid strategy to recover and normally your channels show back up after that if you dont clean the HDD.

adolfosrs, 2019-11-07:
@rootzoll Nice. But do you have any idea on why that happens? Is my SD card getting corrupted or something?

rootzoll, 2019-11-07:
Its always hard to debug such a thing from remote. Dou you use the expensive/pro 32GB sd card like in the shoppinglist? https://www.amazon.de/dp/B07CY3QSST

adolfosrs, 2019-11-09:
@rootzoll Have reflashed the SD card but I'm still having problems. After running the bootstrap script the node reboots and I'm not being able to access it anymore.

> raspiblitz.log

```
***********************************************
Running RaspiBlitz Bootstrap 1.3
Wed  4 Sep 19:38:28 BST 2019
***********************************************
*** Generating new SSH PubKeys
OK
HDD is there but not AutoMounted yet.
Analysing the situation ...
temp-mounting the HDD ..
OK - HDD available under /mnt/hdd
Check if HDD contains already a configuration ..
Found existing configuration
Found valid configuration
Calling Data Migration ..
STARTED Migration/Init
prechecks OK
default values OK
Checking old bitcoin.conf ...
check issue #217 -> ok rpcport exists
check issue #217 -> ok rpcallowip exists
check issue #217 -> ok rpcbind exists
Version Code: 1.3
Version Data: 1.2
detected version change ... starting migration script
TODO: Update Migration check ... only needed after version 1.0
END Migration/Init
Calling Provisioning ..
STARTED Provisioning from preset config file
### BASIC SYSTEM SETTINGS ###
Setting lightning alias: adolfosrs
Auto-Mounting HDD - calling script

*** Adding HDD to the System ***
started from state(recovering)
OK - HDD found as sda

*** Check HDD ***
OK - HDD is formatted with ext4 and is named BLOCKCHAIN
Adding line to /etc/fstab ...
UUID=5a7f971e-6284-4e5a-8cdc-f91af8d66fa2 /mnt/hdd ext4 noexec,defaults 0 2
OK - HDD is listed in /etc/fstab

*** Mount HDD ***
OK - HDD is mounted

tune2fs 1.44.5 (15-Dec-2018)
Setting maximal mount count to 1
moving SSH pub keys to HDD
OK

*** Analysing HDD Content ***
Looks like the HDD is prepared with the Blockchain.
OK HDD got added ... returning to provisioning
Old SSH PubKey exists on HDD > just linking them
Link HDD content for user bitcoin
chown: changing ownership of '/mnt/hdd/bitcoin/chainstate/1187486.ldb': Structure needs cleaning
chown: changing ownership of '/mnt/hdd/bitcoin/chainstate/1187262.ldb': Structure needs cleaning
Copy HDD content for user admin
*** Make backup of LND directory
OK Backup is valid.

*** Adding: raspibolt_67_additional-scripts.md
Creating the command lnbalance as well as lnchannels which will give you a nicer output
OK
installing bash completion for bitcoin-cli and lncli
[... 173 lines cut ...]
Provisioning NETWORK UPnP  - keep default
Provisioning AUTO NAT DISCOVERY - keep default
Provisioning DYNAMIC DNS - keep default
Provisioning RTL - keep default
Provisioning LND Port
User is running standard LND port: 9735
Provisioning DNS Server - keep default
Provisioning Root SSH Keys - keep default
Provisioning SSH Tunnel - not active
Provisioning LCD rotate - not active
Provisioning Touchscreen - not active
Provisioning UPS - not active

*** Replay backup of LND conf/tls
Copying TLS ...
On next final restart admin creds will be updated by _boostrap.sh
DONE

Setting new network hostname 'adolfosrs'
DONE - Give raspi some cool off time after hard building .... 20 secs sleep
END Provisioning
rebooting
Connection to 192.168.0.29 closed by remote host.
Connection to 192.168.0.29 closed.
```

rootzoll, 2019-11-15:
Check this issue #796 .. it could be related to your problem.

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll