raspiblitz/raspiblitz #850: Look for `Bitcoin` folder in 50cloneHDD.sh

openoms, 2019-11-05:
Relayed from Telegram:

>I download the blockchain to an external device and I want to import this blockchain to raspiblitz.
Raspiblitz didn't find the blockchain because it was looking after the folder bitcoin/chainstate.
After renaming the Folder from Bitcoin/chainstate to bitcoin/chainstate it was working.
My idea was as fallback, when raspiblitz can't find bitcoin/chainstate it should altough try Bitcoin/chainstate.

rootzoll, 2019-11-16:
OK - instead of implementing this as a Fallback ... I would do it as a prepare statement. So before the script looks for the "bitcoin" directory .. it will look for a "Bitcoin" dieroctory and if its exists rename it.

rootzoll, 2020-03-16:
The cloneHDD feature is on the fade out. If you have a Blockchain Copy on another computer, you should use the "Copy over Network" option. I know some people use the clone option, but we need to simplify ... if you really want to prepare HDDs with a blockchain - use the CopyStationScript. Closing this issue.

closed 2020-03-16 by rootzoll