raspiblitz/raspiblitz #851: The basic setup is done- Reboot failed

jspeigner, 2019-11-05:
I made to the basic setup is done and rebooted but it never came back up. It says it will take some time to come back up but its been an hour so I restarted it and waited anther 30min

What should I do at this point?

jspeigner, 2019-11-05:
Let it run for another 30min and it came online. So after reboot it could take more than an hour to be even ping-able

rootzoll, 2019-11-07:
Thats unusual .. normally after reboot you can ping your RaspiBlitz very quickly again. Maybe something with your network? What did the LCD says during the time you waited?

jspeigner, 2019-11-10:
It seems that after every reboot it takes my node about 20min to be back online. LCD just flashes a few commands and then goes blank. Maybe its checking the disks after not clean shutdown? But its not ping-able during this time.

rootzoll, 2019-11-15:
Check this issue - it sounds similar: https://github.com/rootzoll/raspiblitz/issues/796 .. working on a solution for v1.4 release. I am closing this issue as a possible duplicate.

closed 2019-11-15 by rootzoll