raspiblitz/raspiblitz #855: Inactive Torrent

martinschultzkristensen, 2019-11-08:
Hi,
After rebooting my Raspiblitz I was told that my blockchain was corrupted.
The system suggested me to re-download torrent. It went well until it turned Inactive. 

___________________________

***` 1) Status Torrent 'blockchain':
  [OPEN]    233718.1 / 233738.1 MB Rate:   0.0 /   0.0 KB Uploaded:     0.0 MB
  Inactive: Hash check on download completion found bad chunks, consider using "

*** 2) Status Torrent 'update':
  [OPEN]    52893.1 / 52907.1 MB Rate:   0.0 /   0.0 KB Uploaded:     0.0 MB
  Inactive: Hash check on download completion found bad chunks, consider using "

___________________________

What does it mean with `consider using "` ???
Other then this my Blitz has been running perfectly since august2019! :)

rootzoll, 2019-11-15:
Not sure how those bad chunks happend (maybe some power probs during download). Yoi can try the https://github.com/rootzoll/raspiblitz#2-copying-from-another-computer option instead of torrent ... that way you even validate the blockchain yourself on your own laptop.

nyxnor, 2021-04-18:
Pls close this.

closed 2021-04-19 by rootzoll