raspiblitz/raspiblitz #856: LND - lnd.service: Main process exited, code=exited, status=1/FAILURE

derPUPE, 2019-11-10:
Dear raspiblitz Wizzards,

i tried to updating an old longtime offline raspiblitz from 1.1 to 1.3
But there is an Problem with reactivation LND. 

 
Debug-log-Paste: https://termbin.com/2ynz

***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.3
chainnetwork: bitcoin / main
 17:32:27 up  9:38,  2 users,  load average: 1.00, 0.80, 0.73

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-11-10 07:56:52 GMT; 9h ago
  Process: 2151 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 2157 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 2208 (bitcoind)
    Tasks: 18 (limit: 2200)
   Memory: 343.3M
   CGroup: /system.slice/bitcoind.service
           └─2208 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Sun 2019-11-10 14:33:55 GMT, end at Sun 2019-11-10 17:32:27 GMT. --
-- No entries --

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2019-11-10T16:41:38Z New outbound peer connected: version: 70015, blocks=603168, peer=665
2019-11-10T16:43:21Z New outbound peer connected: version: 70015, blocks=603168, peer=666
2019-11-10T16:46:47Z New outbound peer connected: version: 70015, blocks=603168, peer=671
2019-11-10T16:50:14Z New outbound peer connected: version: 70015, blocks=603168, peer=674
2019-11-10T16:55:23Z New outbound peer connected: version: 70015, blocks=603168, peer=678
2019-11-10T16:57:07Z New outbound peer connected: version: 70015, blocks=603168, peer=679
2019-11-10T17:02:19Z New outbound peer connected: version: 70015, blocks=603168, peer=683
2019-11-10T17:04:02Z New outbound peer connected: version: 70015, blocks=603168, peer=686
2019-11-10T17:04:29Z Pre-allocating up to position 0x400000 in rev01858.dat
2019-11-10T17:04:30Z UpdateTip: new best=00000000000000000013f69906c90bdda0f02a72f0ca14f784cb841612eee5d6 height=603169 version=0x2000e000 log2_work=91.31804 tx=473351593 date='2019-11-10T17:04:09Z' progress=1.000000 cache=29.0MiB(255021txo) warning='52 of last 100 blocks have unexpected version'
2019-11-10T17:07:28Z New outbound peer connected: version: 70015, blocks=603169, peer=692
2019-11-10T17:12:40Z New outbound peer connected: version: 70015, blocks=603169, peer=698
2019-11-10T17:14:23Z New outbound peer connected: version: 70015, blocks=603169, peer=702
2019-11-10T17:15:30Z socket send error Broken pipe (32)
2019-11-10T17:16:06Z New outbound peer connected: version: 70015, blocks=603169, peer=704
2019-11-10T17:17:46Z UpdateTip: new best=00000000000000000007b4b4ecc637afbc7a70b7012e071958f1739eb145b5f5 height=603170 version=0x20000000 log2_work=91.318066 tx=473354658 date='2019-11-10T17:17:29Z' progress=1.000000 cache=29.7MiB(261325txo) warning='52 of last 100 blocks have unexpected version'
2019-11-10T17:19:39Z New outbound peer connected: version: 70015, blocks=603170, peer=706
2019-11-10T17:21:22Z New outbound peer connected: version: 70015, blocks=603170, peer=709
2019-11-10T17:23:05Z New outbound peer connected: version: 70015, blocks=603170, peer=711
2019-11-10T17:29:04Z Socks5() connect to a63vl5hebrz2cq5j.onion:8333 failed: general failure

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2019-11-10 17:31:46 GMT; 41s ago
  Process: 3353 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
  Process: 3359 ExecStart=/usr/local/bin/lnd --tor.active --tor.v2 --listen=127.0.0.1:9735 (code=exited, status=1/FAILURE)
 Main PID: 3359 (code=exited, status=1/FAILURE)

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Sun 2019-11-10 14:33:55 GMT, end at Sun 2019-11-10 17:32:27 GMT. --
Nov 10 17:30:00 loveandlight4dP systemd[1]: Started LND Lightning Daemon.
Nov 10 17:30:21 loveandlight4dP lnd[915]: Unable to start server: edge not found
[... 55 lines cut ...]
tempFAIL=0
tempWARN=0
tempMAX='5150 centiGrad'

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.0.31'
tempCelsius='45.0'
uptime=34726
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=603170
initialSync=0
syncProgress=100.00 
startcountLightning=421
lndActive=0
scriptRuntime=1

*** OPTION: SHARE THIS DEBUG OUTPUT ***

I tried to backup the channels and restore it another fresh installed raspiblitz, but this does not work. fs Checking the HDD was doen manually. The issue stays the same.

Do you habe any hint form the Debug-log?

Paste: https://termbin.com/2ynz

rootzoll, 2019-11-11:
There is an "edge not found" error why your LND is shutting down. Try to
search for that on the issue list. Its a nasty error from a an old LND
version others have struggled with. Without the channel.backup file ypu may
have to re-setup a new RaspiBlitz and try to recover old cahnnels manually
.. see tutorial by Oliver linked in the FAQs.

nyxnor, 2021-04-18:
Pls close this.

closed 2021-04-19 by rootzoll