raspiblitz/raspiblitz #857: Recovering to new Raspiblitz instance - half of funds in unknown status

nanguapie, 2019-11-10:
This is a continuation of the closed issue https://github.com/rootzoll/raspiblitz/issues/820 which was solved with the help of @openoms who taught me how to best reinstall and recover the raspiblitz. 

My raspiblitz is now running fine with version Raspiblitz 1.3, after I recovered from seed + SCB and thus force closed (at least some of) the channels.

I had 15 channels open when the failed upgrade happened, and it seems that several of them were successfully force closed by my actions when I adhered to this instruction: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-can-i-recover-my-coins-from-a-failing-raspiblitz

When I successfully restored the wallet using seed phrase and the static channel backup file, I received approximately half of my channel funds as on-chain funds due to closing of approx half of the channels.  Interestingly, 6 channels are still in the status "Waiting Close Channels (6)" when looking in the RTL interface, and they have been so since the restoration (1-2 weeks ago now) 

**ISSUE 1)**  I had a bit of on-chain sats as well, and those have not yet been restored. the on-chain wallet started at 0 sats, and then increased to approx half my channel funds after the force closing. I believe I will need to use another wallet implementation and restore from seed to get the on-chain lost funds. I tried this now with the LND alpha version for Mac OS, but it didn't accept my seed encryption password (which did work using the RaspiBlitz restoration process).  

**ISSUE 2)** Very interestingly,  the 6 channels that are in the "Waiting Close Channels" status can be found on 1ML.com and it is visible there that the channels are still open. However, using the command line I cannot see that I have any channels open or even in pending status. They are only visible in the Ride The Lightning web-interface and on 1ML.com.  

Is there any way that I can force close (or coop close) these channels now, which are connected to my node, but not acknowledged by my node?  Or will I have to wait until months/years in the future where the other side may finally decide to force close the channels and then potentially receive my part? 

Thank you for any advice!  I can provide more details regarding the exact channels that should be closed if that would help.

nanguapie, 2019-11-10:
Correction - using the command "lncli pendingchannels" I can see the 6 channels that are listed in the RTL interface as well. And I can get some more details using lncli getchaninfo --chan_id .......

However, I do not know how to close them, thankful for guidance!

rootzoll, 2019-11-15:
Not sure if its possible to unpack every problem in here ... but give it a try.

Issue1) The recover of on-chain funds is as described sometimes strange on RaspiBlitz and thats why it recommend to do it on the LND Desktop like you tried. But there is that Password problem .. if you protected your seeds with passwords the LND Desktop has a UX prob with that sadly .. thats why lastest RaspiBlitz is not asling you anymore fro password encrypting your seed to prevent this situation. If you come from an older RaspiBlitz I dont have a solution for that other then to bug the people over at the LND Dev Slack what to do: https://dev.lightning.community

Issue2) I am not sure what is the channel listing policy is on 1ML.com

nanguapie, 2019-11-25:
Thank you very much for your feedback @rootzoll , much appreciated. 

I understand your explanation of issue1), and i will bug the LND guys lightly regarding this UX issue, I'm sure it'll be improved soon and possible easily to recover this on-chain part of my funds.  (probably it already would work in other lightning apps, i just haven't tried yet) 

issue2) is a bit more unclear to me. I understand that it's not possible to deduce very much from the 1ML.com channel listing, but I can still see in my own online RTL interface that the 6 channels with funds are stuck in some kind of limbo state -"waiting close channel"- .  And I wonder if there is any action i can take to force close them and of course take those on-chain fees which might apply at the time of close.  Please let me know if there is any action to be taken, or if there is any rule behind how long time it should take before the channels are closed.  I thought that all channels would be force closed by me restoring the channel backup file, but apparantly only approx half of the channels were actually closed and funds returned.  And during the weeks since I got the node back up and running, no change has occured in the list of the 6 "waiting close channel" channels. 

Thanks again, hope this scenario provide some interesting feedback and maybe someone here have a view of possible actions to take. Perhaps this is more interesting for the LND crew than for RaspiBlitz though I start to suspect..

openoms, 2019-11-25:
Restoring from.the SCB can only trigger the force close when your peers are online.
Of you can contact you peers with the pending channels ask them to force close.
Otherwise you need to either wait until they come online or attempt to restore with the SCB again if they came online since the last attempt and the channels still pending.

nanguapie, 2019-11-25:
Thank you @openoms , I didn't realize that force close required peers to be online. 

If I understand you right, even restoring with the SCB again would also require them to be online.  What would happen if they have tossed their node in the trash and never come back again? Would that be one of the garbage dump donations to the bitcoin community in the form of deflation or would any type of timelock eventually move the funds?  Seems like some kind of way to get funds out of this situation would be required to build for the future if it's not there as of now. 

I can also try to contact the node operators based on 1ML info, but don't know them personally. I mean it's not a big deal, but a sat saved is a sat earned :) I will make sure to send a part of them to you and rootzoll if i get them back in the end!

openoms, 2019-11-25:
You got it right. As for now if both partys lose the information of the channel states (also by not coming back online) the funds will be locked in the multisig contract forever.
Saying that if your peer is permanently offline you don't need to worry about restoring an old state (any old channel.db) because they can't punish you with broadcasting the justice transaction taking all the funds in the channel (except if there is a watchtower set up and watching to do it for them).

Unfortunately this is likely to remain the state of LN until the opcode: sighash_noinput is merged into Botcoin Core and makes ELTOO (a different protocol for storing LN states and doing justice transactions) will be introduced.

closed 2019-11-26 by rootzoll