raspiblitz/raspiblitz #860: SSH "Connection refused" after reboot

esprograms, 2019-11-13:
I successfully went through all the steps up to "Final LND Setup" in the setup documentation (https://github.com/rootzoll/raspiblitz/#setup-process-detailed-documentation). I have a RaspberryPi 4 with 4GB of RAM. I selected to run the node behind TOR and to download the blockchain using the Sync method. I was expecting this to take a really long time, but it almost just skipped on to the LND Setup, so I set up a new wallet and obtained my 24 seed words. I then continued, bringing me to this stage:

[image]

After rebooting, I got this screen on my RaspiBlitz:

[image]

When then trying to SSH, I get the error "ssh: connect to host ###.###.##.### port 22: Connection refused."

The only thing I can think of is that maybe the blockchain is currently downloading in the background, using the full network.

Any help is appreciated.

Thanks

esprograms, 2019-11-13:
I should have noted that after clicking on the "Ok" to reboot 

[image]

I received this message in my command line: 

"Setup done. Rebooting now.
error: Could not locate RPC credentials. No authentication cookie could be found, and RPC password is not set.  See -rpcpassword and -stdinrpcpass.  Configuration file: (/home/bitcoin/.bitcoin/bitcoin.conf)
Connection to ###.###.##.### closed by remote host.
Connection to ###.###.##.### closed."

frandani82, 2019-11-13:
I have exactly the same issues. Tried to build my raspiblitz (Raspberry Pi 4 4GB) yesterday and everythings worked fine until the reboot. After the reboot I can't connect to the raspiblitz anymore.

Xorell, 2019-11-13:
I am facing the same issue, I am working on Raspberry Pi 4 4GB, HDD MX500 1TB and use the official raspberry PI 4 supply.

I was doubtful about the supply an therefore i swapped it with my phone charger, which can supply less power (only 2A) but strangely i could login through the SSH. 

I have tried this several times and I could only restore the SSH with my phone charger. If i go back to the official power supply afterwards then the SSH is active again (but not always from the first attempt), not clear if it is luck or if the phone charger changes something in the behavior of my PI.

rootzoll, 2019-11-15:
@esprograms if its downloading the blockchain you should see a progress status and SSH login should work. So not exactly sure whats going on. We had some reports that either the SSH needs a long time to boot up because of the plymouth packages: An idea would be to start with a fresh sd card and try to remove that packages on first login like mentioned here:
https://github.com/rootzoll/raspiblitz/issues/796#issuecomment-552233584
or wait for the v1.4 release where this should get fixed out of the box.

Also you can try to change the power supply like mentioned in that issue link above.

@Xorell beside the Ampere a stable voltage supply is also important - maybe your phone charging power supply is handeling this better. At the moment we recommend the following power supply thats seems to run best in tests: https://shop.pimoroni.com/products/universal-usb-c-power-supply-5-1v-3a

frandani82, 2019-11-16:
I tried it again with flashing a new SD Card. After the first boot up I removed plymouth with
sudo apt-get purge plymouth*

Then I made the complete installation and everything went fine
[image]

after the boot up I got this message
[image]

I could continue with CTRL D but the in showed me a error. I have this log file:

* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.3
chainnetwork: bitcoin / main
 08:02:13 up  7:47,  2 users,  load average: 1.90, 1.66, 1.71

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2019-11-16 08:01:54 GMT; 19s ago
  Process: 12446 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 12448 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=1/FAILURE)

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Fri 2019-11-15 23:31:42 GMT, end at Sat 2019-11-16 08:02:14 GMT. --
Nov 16 08:01:54 raspiblitz systemd[1]: bitcoind.service: Service RestartSec=30s expired, scheduling restart.
Nov 16 08:01:54 raspiblitz systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 564.
Nov 16 08:01:54 raspiblitz systemd[1]: Stopped Bitcoin daemon.
Nov 16 08:01:54 raspiblitz systemd[1]: Starting Bitcoin daemon...
Nov 16 08:01:54 raspiblitz bitcoind[12448]: Error: Cannot write to data directory '/mnt/hdd/bitcoin'; check permissions.
Nov 16 08:01:54 raspiblitz systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE
Nov 16 08:01:54 raspiblitz systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Nov 16 08:01:54 raspiblitz systemd[1]: Failed to start Bitcoin daemon.
1573891307 ERROR: Error found in Logs
1573891328 ERROR: Error found in Logs

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2019-11-16T03:10:03Z UpdateTip: new best=00000000000000001ae150a5efdd4e6bccba8b5eec3578a1967deccba728cf09 height=300226 version=0x00000002 log2_work=78.525536 tx=38531504 date='2014-05-11T13:30:19Z' progress=0.081923 cache=165.8MiB(1001062txo)
2019-11-16T03:10:03Z UpdateTip: new best=00000000000000001ea3e01af95e3005e63262174973a521c7637fdcb210955f height=300227 version=0x00000002 log2_work=78.52565 tx=38531571 date='2014-05-11T13:30:14Z' progress=0.081923 cache=165.8MiB(1001711txo)
2019-11-16T03:10:03Z UpdateTip: new best=000000000000000050ee7169975aea2c169df3c34a8e09d1956915613c7495d3 height=300228 version=0x00000002 log2_work=78.525764 tx=38531815 date='2014-05-11T13:35:45Z' progress=0.081923 cache=165.9MiB(1001745txo)
2019-11-16T03:10:04Z UpdateTip: new best=000000000000000071b7d470fb9cd9b046c4749f4665611e4bf4697ee15f6427 height=300229 version=0x00000002 log2_work=78.525878 tx=38531870 date='2014-05-11T13:34:19Z' progress=0.081923 cache=165.9MiB(1001971txo)
2019-11-16T03:10:54Z UpdateTip: new best=0000000000000000248602b7db925e01dd7e9f589c83647286db8857d41f8cb8 height=300230 version=0x00000002 log2_work=78.525992 tx=38532182 date='2014-05-11T13:49:08Z' progress=0.081924 cache=166.0MiB(1002854txo)
2019-11-16T03:11:29Z Pre-allocating up to position 0x400000 in rev00139.dat
2019-11-16T03:11:29Z UpdateTip: new best=00000000000000004310299b01861f0de0c50c4cf7a0c8ff63cbe7796987fa4f height=300231 version=0x00000002 log2_work=78.526105 tx=38533064 date='2014-05-11T14:13:39Z' progress=0.081926 cache=166.1MiB(1004464txo)
2019-11-16T03:11:30Z UpdateTip: new best=000000000000000062fd025c310484520273bf08ce8a7db72f9a5b8705677f59 height=300232 version=0x00000002 log2_work=78.526219 tx=38533245 date='2014-05-11T14:21:22Z' progress=0.081926 cache=166.2MiB(1004595txo)
2019-11-16T03:13:59Z UpdateTip: new best=00000000000000004993beea05a04ac593214a4b397900e7b69e49d9cd46f12a height=300233 version=0x00000002 log2_work=78.526333 tx=38533347 date='2014-05-11T14:20:23Z' progress=0.081926 cache=166.2MiB(1004640txo)
2019-11-16T03:13:59Z UpdateTip: new best=00000000000000004ddb2439cbb949571dcf49e3dcbc40115ddc0880aee0b048 height=300234 version=0x00000002 log2_work=78.526447 tx=38533402 date='2014-05-11T14:19:56Z' progress=0.081926 cache=166.2MiB(1004722txo)
2019-11-16T03:13:59Z Pre-allocating up to position 0x3000000 in blk00139.dat
2019-11-16T03:13:59Z UpdateTip: new best=000000000000000076201477c2073a998a0bc3d0a35e91dbf16ae9ab27be1aee height=300235 version=0x00000002 log2_work=78.526561 tx=38533614 date='2014-05-11T14:25:53Z' progress=0.081927 cache=166.2MiB(1004858txo)
2019-11-16T03:13:59Z UpdateTip: new best=00000000000000004e7f691138b840097adc3c278452ea7617a8b96839a221bd height=300236 version=0x00000002 log2_work=78.526675 tx=38533870 date='2014-05-11T14:29:42Z' progress=0.081927 cache=166.2MiB(1004981txo)
2019-11-16T03:13:59Z UpdateTip: new best=000000000000000050fd4f82924db77d1381a68da60b4f2cf106b688979a6ff1 height=300237 version=0x00000002 log2_work=78.526789 tx=38533934 date='2014-05-11T14:55:53Z' progress=0.081928 cache=166.2MiB(1005126txo)
2019-11-16T03:13:59Z UpdateTip: new best=00000000000000004aba93a2eb1121e6d5073c67f4f8389ac4205a474bd448a8 height=300238 version=0x00000002 log2_work=78.526903 tx=38534838 date='2014-05-11T15:03:30Z' progress=0.081930 cache=166.3MiB(1005682txo)
[... cut ...]
# OK: log 'ERROR: Error found in Logs' written to /home/admin/systemd.blockchain.log
startcountLightning=1
lndActive=1
/home/admin/config.scripts/blitz.statusscan.sh: line 130: /mnt/hdd/temp/.lnd.error: Read-only file system
lndRPCReady=0
walletOpened=0
walletReady=0
walletLocked=1
lndRPCReady=0
scriptRuntime=1

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

My SDD ist the Cruicial SSD MX500. Any ideas?

rootzoll, 2019-11-16:
The logs say:  "Cannot write to data directory '/mnt/hdd/bitcoin'" ... so something is messed up with the writing rights to your HDD. This can be because of those "Maintance Warning" bootup.

Normally when you get the "Maintance Warning" I recommend to make a fresh sd card image and run thru the automatic update/recover process and then this is fixed. I know you just did that, but can you give it one more try and removing the plymouth then on first login just again?

The SSD model looks good. What Power Supply are you using?

frandani82, 2019-11-16:
I use this Power Supply:
https://www.pi-shop.ch/netzteil-fuer-raspberry-pi-4-model-b-usb-c-5-1v-3a-uk-eu

I am not sure if it is a Problem with the hard drive. I made a new SD Card and even deleted the Hard Drive.
[image]

It seems that everthing is fine with the HDD but when Bitcoind is starting this appears
[image]

and 10 sec later
[image]

If I restart the I boots in the emergency mode as describe above in this post.

rootzoll, 2019-11-25:
Your power supply looks good. What HDD adapter/case are you using? There was an issue with some HDD adapter/cases that dont support UASP fully and freezing during use. The case/adapter you use should explicitly say that it supports UASP like this one from the shopping list: https://www.amazon.de/gp/product/B06XWSDGP6

frandani82, 2019-11-26:
the adapter is this one
https://www.brack.ch/icy-box-externes-gehaeuse-ib-ac703-u3-2-5-828346?query=icy
UASP Support should be ok.

geco91, 2019-11-26:
Seems like this bug is is hard to get, see if we can work around it  ... it looks like as you choose to self sync the chain with the pi ... did you try the alternative COPY option to validate the blockchain on your laptop and then copy it over?

frandani82, 2019-12-04:
I also tried it with a torrent download and copy a harddisk. both doesn't work. With an other Harddisk (1TB Harddrive) it works. I tried also the SSD with an other Bitcoin Node (mynode). That works, so it can not be a problem with the SSD. I have no idea what else to try.

sicXnull, 2019-12-15:
i dealt with this problem for hours. Plug your HDD into a linux machine so you can see the filesystem. Look and see if there is a "SSH" folder. If so, delete the keys inside that folder.(leave all the other config files).In mine, there was a SSH folder inside the SSH folder. do the same thing there. 

Then remove and plug back into pi. 

You should be able to SSH now.

raspito, 2020-06-18:
I believe I am also experiencing this / #1042 after a fresh install. Once the install script finishes creating the lightning wallet and does the final reboot (for the screen drivers - note: I don't have a screen) I get "connection refused" when I go to log back in. Will follow the steps described here, but it looks like something is going wrong during that script.
https://github.com/rootzoll/raspiblitz/blob/v1.5/FAQ.md#how-can-i-repair-my-ssh-login

raspito, 2020-06-18:
Tried ssh.reset and ssh - neither worked. Will try to build from scratch. Is it possible that this could have something to do with the screen drivers? Perhaps we should add a Y/N prompt depending on whether the screen exists?

rootzoll, 2020-06-18:
> Is it possible that this could have something to do with the screen drivers? Perhaps we should add a Y/N prompt depending on whether the screen exists?

It would be good to install screen drivers just when really needed ... but this would need to be a automatic detection if the LCD Hat is connected or not ... also for other reasons.

I created a issue for this research topic: #1265

Xorell, 2020-06-18:
> Tried ssh.reset and ssh - neither worked. Will try to build from scratch. Is it possible that this could have something to do with the screen drivers? Perhaps we should add a Y/N prompt depending on whether the screen exists?

IF the problem persist i would try to go through the installation process with a powered USB hub. For me it seemed to be the solution, even after trying out different supplies. Once installed you should be fine without the hub ( not fully sure, I tried it and it was fine but today I am using a hub to connect other hardware)

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll

dskvr, 2022-12-19:
Error still exists with most recent version, using approved hw.

rootzoll, 2022-12-21:
Does at least the WebUI work for you? with calling in browser: http://[LOCAL_IP_OF_RASPIBLITZ] ?

dr2okevin, 2024-12-02:
I have the same Problem with a Raspi5 and raspiblitz-fat-v1.11.3-2024-11-03.img.gz
I copy-pasted the password from a text file during setup via SSH. After reboot I get connection refused before I can even type any password. Web UI says wrong password.

rootzoll, 2024-12-03:
When u get wrong password the only way to reset ist is to flash a new image to the sd card and to an update/recovery again where you can set the passwordA fresh.