raspiblitz/raspiblitz #862: Heatsink tests the RPi4

openoms, 2019-11-16:
Starting with this comment: https://github.com/rootzoll/raspiblitz/issues/796#issuecomment-554368326
I wanted to test some heatsinks to achieve a no-moving parts setup for the RPi4.

Having no good experience so far with this model (without a fan): https://www.aliexpress.com/item/33046089229.html

Managed to drive up the temperature to 80 degrees and rising when running the hardware test for 5 mins.
The thermal connection of the heatsink case was a bit questionable due to the supplied pads being just too thin. During the test they proved to work as the whole blob of aluminium got hot. The LCD screen is sitting just on top, that I found worrying at >80 degrees so stopped.

A heatsink with fans would be a option (fan could be connected to the pins of an LCD-with-power-pins), but it has the drawback of being always on, most of the time unnecessarily.

In my opinion the RPi4 still needs active cooling during high power tasks like the Initial Block Download. For this the Shim Fan perfect because can automatically switch on for higher temperatures and sit silent when within a comfortable threshold (I have set <70 degrees - completely arbitrarily, just because it is sitting silent on my desk during the day-to-day operation)

rootzoll, 2019-11-20:
OK .. so lets stick to the ShimFan for now. Thanks.

osilloh, 2019-11-29:
First photo
First Pi image top left passive cooler is BTCpay server Rpi4, has 3 channels open. Old firmware.
Second Pi middle is a Pi3 running minimal load, 2 channels.
3rd Pi in Crypto case, the Fanshim inside died, it was running around 70C with almost no load. No channels open. I noticed the temperature much higher at 87C, the Fanshim died. Old firmware.

Thermal photo:

Notice the heat sink on the Rpi4 on top is about 111F ~ 40C (the scale is the color on the left hand side, white being the hottest)
2nd Pi has the screen, which blocks the heat sources.
3rd Pi also has screen and case obscuring the thermal camera.

Third photo:

Replaced dead Fanshim(lasted about a month in the Crypto case) with passive heatsink (like the one in the top Rpi4, but bottom plate could not fit in case), updated the firmware to:  000137ab. Not sure if the firmware is keeping the clorckrate/temperature down, or the passive sink is doing its job. Perhaps I should have only replaced the Fanshim as a better test. In any event, it is running cooler as you can see, 68C or 154F. Which is close to the 70C it used to run with the Fanshim. This Rpi4 in the case is running a minimal load. The passive sink compared to the dead Fanshim is about 19C lower. I prefer to run a passive heatsink at the moment.

Hopefully someone has the time/resources to run a better comparison. This is a real swag, but thought it interesting to post.

[image]
[image]

[image]

openoms, 2020-02-15:
From TG:
[image]

openoms, 2020-02-15:
Heatsink cases don`t seem to be viable solution without a fan, but listing them anyway:

* Heatsink case with GPIO exposed (needs testing how the LCD takes the constant warmth underneath)
[image]
https://www.amazon.com/Geekworm-Raspberry-Ultra-Thin-Aluminum-Compatible/dp/B07X5Y81C6/
or
[image]
https://www.amazon.com/Geekworm-Raspberry-Computer-Aluminum-Compatible/dp/B07VD568FB/

amcitaly, 2020-03-13:
@openoms hello, I would like to know if you have tried the heatsink with fan, like the one show in the pics of Aliexpress... I would like to buy one in amazon to give a try, but I don't know if then will make problem as the Pimoroni shim fan. Thank you

openoms, 2020-03-15:
@amcitaly. The shimfan has an extra PCB causing problems, other fans needing only power won't have that.

The problem with the heatsinkcase eith fan is that it is blocking the pins needed by the LCD, but you might be able find a screen with power pins and test with that. See: https://github.com/rootzoll/raspiblitz/issues/1006#issuecomment-585778441

rootzoll, 2020-03-16:
@amcitaly I am testing the heatsink-case-WITHOUT-fan at the moment ... during self-validation of the blockchain I get up to 77.4 Celsius with a 68 Celsius average .. so it gets very hot - but at it looks like I will get thru the complete self-validation within 3 days - what is not bad and maybe an acceptable slowdown. Maybe this can even be optimized by placing distance holders between SSD & Pi-Heatcase. 

I have to see how it compares with a active fan once it is fully synced and CPU is running not that 100% ... but so far at least I can say that the heat-sink-case (without fan) can be used as an temporarly replacement for the shim fan.

amcitaly, 2020-03-16:
@rootzoll Thank you, in fact I have always used the heatsink case without fan on the Rpi3 and was good, it was my ultimately case, not attached to the HDD but separated. I want to buy the new one for the Rpi4 on Pimoroni (I just waiting the colored blue or violet and now they are out of stock : ) 
I will buy it for sure while we find a good fan.

3 days of validation is ok, with my Rpi3 on torrent it took more or less one week to fully sync, and maybe once the blockchain is validated it will get even cooler.