raspiblitz/raspiblitz #869: RaspiBlitz at IPv6 only

PatrickScheich, 2019-11-19:
Is there a way to (re)configure an existing/already running RaspiBlitz to an IPv6 only configuration?

IPv4 wise I'm behind an ISP NAT so I get an external IPv4 address but will never be reachable by this.
I already configured my RaspiBlitz to use IPv6 and a DynDNS (also IPv6 only), but my Bitcoin- and Lightning-Node is only listed by the (unreachable) IPv4 address.

frennkie, 2020-08-08:
The idea is to go with TOR-by-default. IPv6 is would kind-of be the opposite direction. 

Do you still feel a need for this?

PatrickScheich, 2020-08-09:
> 
> 
> The idea is to go with TOR-by-default. IPv6 is would kind-of be the opposite direction.
ok, I had not looked into the TOR option yet

But as I "suffer" from the CGNat my ISP is giving me I had recently looked into the raspiblitz source code during my update to v1.6.0.

I have a "hardcoded" solution which is surely not desirable but may be useful for reference 

Perhaps a Setting/Option may be feasible

I would like to contribute but I'm not very keen with the way to collaborate on a public github project
So I may need some advice here

rootzoll, 2020-08-09:
@PatrickScheich If its just some additional/modified lines just paste them here as example and then we can see how complex it would be to add that as an option.

PatrickScheich, 2020-08-11:
> 
> 
> @PatrickScheich If its just some additional/modified lines just paste them here as example and then we can see how complex it would be to add that as an option.

great, I attached a Zip containing the 4 files with changes
[IPv6 changes.zip](https://github.com/rootzoll/raspiblitz/files/5057374/IPv6.changes.zip)

- _background.sh
- _bootstrap.sh
- 00infoBlitz.sh
- config.scripts/bonus.electrs.sh

basically I changed 

```
freshPublicIP=$(curl -s http://v4.ipv6-test.com/api/myip.php 2>/dev/null)
````

by 

```
freshPublicIP=$(curl -s http://v6.ipv6-test.com/api/myip.php 2>/dev/null)
```
and added the square brackets around the new IPv6 address as this is needed in `lnd.service` 

unfortunately there are usages of `publicIP` with the netcat tool which does not like the square backets, so I introduces an additional (internal) variable to get an un-brackified-version at those points.

rootzoll, 2020-08-17:
@PatrickScheich will take a look at your proposed changes and see if we can use them for v1.6.1 in any kind of way.

rootzoll, 2020-09-17:
@PatrickScheich I added your changes, moved a lot of them into the `internet.sh` script. You can now test them by updating your v1.6 RaspiBlitz with: MENU > UPADTE > PATCH > PATCH-MENU > PATCH

Then you can on terminal switch on IPv6 with `./config.scripts/internet.sh ipv6 on` (make sure TOR is off and reboot). Then you should see that the Bitcoin & LND pickup the IPv6 address.

PatrickScheich, 2020-09-23:
> @PatrickScheich I added your changes, moved a lot of them into the `internet.sh` script. You can now test them by updating your v1.6 RaspiBlitz with: MENU > UPADTE > PATCH > PATCH-MENU > PATCH
>
> Then [...]

@rootzoll thanks, I have done as proposed but `/home/admin/_background.sh` seems broken
it terminates with a syntax error
```
Sep 23 14:47:39 DesertRetreatLN _background.sh[7910]: _background.sh STARTED
Sep 23 14:47:39 DesertRetreatLN _background.sh[7910]: /home/admin/_background.sh: line 139: syntax error near unexpected token `fi'
Sep 23 14:47:39 DesertRetreatLN _background.sh[7910]: /home/admin/_background.sh: line 139: `  fi'
Sep 23 14:47:39 DesertRetreatLN systemd[1]: background.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Sep 23 14:47:39 DesertRetreatLN systemd[1]: background.service: Failed with result 'exit-code'.
Sep 23 14:47:49 DesertRetreatLN systemd[1]: background.service: Service RestartSec=10s expired, scheduling restart.
Sep 23 14:47:49 DesertRetreatLN systemd[1]: background.service: Scheduled restart job, restart counter is at 17760.
Sep 23 14:47:49 DesertRetreatLN systemd[1]: Stopped RaspiBlitz Background Monitoring Service.
Sep 23 14:47:49 DesertRetreatLN systemd[1]: Started RaspiBlitz Background Monitoring Service.
Sep 23 14:47:49 DesertRetreatLN _background.sh[8323]: _background.sh STARTED
Sep 23 14:47:49 DesertRetreatLN _background.sh[8323]: /home/admin/_background.sh: line 139: syntax error near unexpected token `fi'
Sep 23 14:47:49 DesertRetreatLN _background.sh[8323]: /home/admin/_background.sh: line 139: `  fi'
Sep 23 14:47:49 DesertRetreatLN systemd[1]: background.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Sep 23 14:47:49 DesertRetreatLN systemd[1]: background.service: Failed with result 'exit-code'.
Sep 23 14:47:59 DesertRetreatLN systemd[1]: background.service: Service RestartSec=10s expired, scheduling restart.
Sep 23 14:47:59 DesertRetreatLN systemd[1]: background.service: Scheduled restart job, restart counter is at 17761.
Sep 23 14:47:59 DesertRetreatLN systemd[1]: Stopped RaspiBlitz Background Monitoring Service.
Sep 23 14:47:59 DesertRetreatLN systemd[1]: Started RaspiBlitz Background Monitoring Service.
Sep 23 14:47:59 DesertRetreatLN _background.sh[8587]: _background.sh STARTED
```

It turns out that there is an `else...fi` block which causes this syntax error, lines 135 to 137.
I attached the corrected _background.sh for your reference
[_background.sh.txt](https://github.com/rootzoll/raspiblitz/files/5272012/_background.sh.txt)

rootzoll, 2020-09-24:
@PatrickScheich thanks .. fix is now merged into v1.6 brach. You can patch update if you want to continue testing.

rootzoll, 2020-10-08:
OK not able to confirm the IPv6 but changes seem not breaking any IPv4 environment. So this will be part of v1.6.1 release. Closing issue.

closed 2020-10-08 by rootzoll