raspiblitz/raspiblitz #870: Touch UI test - Err: LND is locked

openoms, 2019-11-19:
Built a new SD-card manually from master on my RPI4 and restored a previous blockchaindata and LND directory.
After activating finishing syncing and activating the touchscreen I have an error message when pressing the ` Node` or `Invoice` button saying:
 `Err: LND is locked`

LND is unlocked manually (no autounlock currently) and works fine.
`lncli getinfo` works, gives the full output.

@frennkie or anyone can you reproduce this?

frennkie, 2019-11-19:
Can you check the log for hints: /home/pi/blitz-tui.log

openoms, 2019-11-19:
Nice, it has a log! 
Could not make much out of it though. 

Here is the full log:
`$ cat /home/pi/blitz-tui.log | nc termbin.com 9999`

https://termbin.com/w68f

openoms, 2019-11-19:
Now uninstalled and reinstalled from the `SERVICE` menu.
Initially it showed the `Node` URI, but gave an error about `channels not open` despite 3/3 were active on the status  screen.
Couple of seconds and it all works!
Recorded the second logs of the second install too:
`$ cat /home/pi/blitz-tui.log | nc termbin.com 9999`
https://termbin.com/dyge

I attribute the problem to the fact that I have migrated to a backed up LND directory.
It seems the reinstall sorted the problem. 

Thank you for your input, being able to look at the specific logs helps a lot!
Will close this now.

closed 2019-11-19 by openoms