raspiblitz/raspiblitz #872: "The bitcoind service is not running. Login for more details" error

smallroy21, 2019-11-21:
I'm getting the "The bitcoind service is not running. Login for more details" error during the blockchain syncing step. Restarting my raspi worked temporarily but is no longer helping.

I saw another thread say to start bitcoind manually but i dont know the command for that.

[bug logs](https://pastebin.com/jDYPVkx2)

Thank you

Dikay900, 2019-11-22:
just start the mainmenu again after some minutes.
from your logs you can see that all services are running and your bitcoin debug log shows the blocks are getting forwarded.

Got the same message just now and could restart the mainmenu with ./00mainmenu.sh after waiting some time.

smallroy21, 2019-11-22:
I can get it to the main menu fine but how do i get it to go back to continue syncing the blockchain?

Dikay900, 2019-11-22:
you can use the following command to see if your syncing:

sudo tail -f /mnt/hdd/bitcoin/debug.log

2019-11-22T19:28:09Z UpdateTip: new best=00000000000000000014e9658245c3c9eae84499b3d8b3825b49d8d54a178c08 height=604366 version=0x20000000 log2_work=91.348288 tx=475791175 date='2019-11-18T14:06:34Z' progress=0.997181 cache=43.7MiB(392196txo)

If there are new lines popping up then you just have to wait for progress to get to 1.00 which is reached when you are fully synced

nyxnor, 2021-04-18:
Pls close this.

closed 2021-04-19 by rootzoll