raspiblitz/raspiblitz #881: Project inclusion to Raspiblitz

fusion44, 2019-11-25:
Hey all,

I'm working on yet another mobile LN wallet, but I'd like to pivot a little to make it more a frontend for full nodes. Raspiblitz would be perfect for that. 

Here's a video of what I'm talking about:
https://www.youtube.com/watch?v=udfQZoITn9Q

Here you can see my app running on Desktop. The last few seconds capture a Raspberry PI with a touchscreen. If you see red elements, it is because some an error was thrown. If it looks a bit choppy it is because the app is running in debug mode - release mode should be smooth. Lots of things remain to do to make it look more beautiful on Desktop since I just run the Phone app on the Desktop. Flutter runs on the PI without a window manager like X as it talks directly to the graphics chip.

That said, Flutter on Desktop (or on a PI for that matter) is still a work in progress, and some things are still missing. I was surprised that I got it working this smoothly. In theory, the app should also compile as a web app, but I think there are many excellent options so I won't optimize for that.

Source: https://github.com/fusion44/sendmany
Tech: https://flutter.dev/
License: MIT

Implemented features: 
- [x] Create invoices
- [x] Send payments
- [x] Send and receive on-chain transactions 
- [x] Channel management (currently only opens channels, closing channels is todo)
- [x] i18n support (currently available: English, German and amateurish Norwegian support)
- [x] Theme support
- [x] Multi remote LND support
- [x] Runs on phones (currently only optimized for Android)

Future features I'd like to implement are more UI polish, a block explorer using electrum personal server and LN network explorer. I aim more for the power user as the target audience as I don't hide complexities.

My goal is to include it with Raspiblitz as an option in the long run. I'd even be open to rebrand the app to be more closely aligned to the Raspiblitz project. What do you think?

openoms, 2019-11-25:
It is a very nice project. Reminds me slightly to the philosophy of Zeus (which is my favourite), but optimized for the desktop.

Aiming to not hide complexities does align very well with the RaspiBlitz.

Regarding the block explorer I have just merged the support for BTC-RPC-Explorer (which supporst using ElectrumX and ElectRS for adress look up)

Electrum Personal Server is not suitable for this, because only tracks the preser addresses, but you might just referred to an Electrum Server there.

An own LN explorer is great. RTL has some limited function for stats based on the nodes own view and peer lookup, I am not sure if they are planning to expand that.

There was a discussion before about a BlitzDaemon (mentioned by either @frennkie or @emzy), not sure where that project stands. The idea was to give an API to the RaspiBlitz to be able to communicate the settings and stats  (some of which are already displayed on the status screen). Once that would progress a specific desktop/mobile companion app would make very much sense.

fusion44, 2019-11-26:
Thanks :)

I might be able to help with the aspect for the server. Some time ago, I wrote a server component which manages LND instances for each user registered. It spins up an LND instance whenever a new user registers. The server provides a thin GraphQL abstraction layer to LND to make the API more mobile-friendly (mainly to save bandwidth and CPU cycles on phones). Though, this is probably total overkill for the Raspiblitz :)
source: https://github.com/fusion44/fort-bitcoin-gql

openoms, 2019-11-28:
Found the plans for the infoservice, there are things written  already, have a look: https://github.com/rootzoll/raspiblitz/issues/88

openoms, 2019-12-13:
Great to see this, well played: https://twitter.com/joostjgr/status/1205233172319002625

nyxnor, 2021-04-18:
It is comming, well, since 25/11/2019. Gonna be good.

rootzoll, 2021-04-27:
Hi. I am now sorting options fo the v1.8 new WebUI & reworked TouchUI .. also of course the Backend options that come with that. @fusion44 can you give an update on whats the status of the sendmany RaspiBlitz inclusion?

fusion44, 2021-04-28:
I've posted a few thoughts on this in issue #43