raspiblitz/raspiblitz #883: Can not connect from Zap after running RESET-LND and restored a node from seed

adolfosrs, 2019-11-26:
I have reset my LND node and restored a node from a seed phrase. 
I was able to connect to the previous node using Zap but the new one is not reachable. I did regenerate the QR code and rescanned it from zap but I'm getting `Could not connect to server (Connect Failed)`.

Am I missing anything?

Thanks

geco91, 2019-11-26:
Which ZAP are you using iOS or Android? Did you used the "reset connection" within ZAP before scanning the new QR code?

adolfosrs, 2019-11-26:
iOS

I have totally removed the old node configuration from the app. All I see now is the screen with the **Scan** and **Paste** button.

openoms, 2019-11-26:
You will need to reset your macaroons and TLS cert from the EXPORT menu and pair Zap again.

adolfosrs, 2019-11-26:
@openoms Thanks but it did not work. The same scenario is still happening.

openoms, 2019-11-27:
There is more up to date info here: https://github.com/openoms/bitcoin-tutorials/blob/master/Zap_to_RaspiBlitz_through_Tor.md

You can run lndconnect manually:
add it to PATH:
```
export GOROOT=/usr/local/go
export PATH=$PATH:$GOROOT/bin
export GOPATH=/usr/local/gocode
export PATH=$PATH:$GOPATH/bin
# make the path persist
sudo bash -c "echo 'PATH=\$PATH:/usr/local/gocode/bin/' >> /etc/profile"
```

and run 
`lndconnect`
 or
 `lndconnect -j` or the text string for Zap desktop.

new lndconnect version (should not make a difference unless you are trying to connect through Tor):
```
# Install lndconnect from source:
go get -d github.com/LN-Zap/lndconnect
cd $GOPATH/src/github.com/LN-Zap/lndconnect
make
```

Are you using a DynDNS? Switch it off and try again or try using Tor if on iOS

adolfosrs, 2019-11-27:
>Are you using a DynDNS? Switch it off and try again

**Switching it off and turning it on again did the trick.** 

Thanks!

closed 2019-11-27 by adolfosrs