raspiblitz/raspiblitz #885: Manual SD card build problems

openoms, 2019-11-27:
Following the procedure here: https://github.com/rootzoll/raspiblitz#build-the-sd-card-image
with the latest Raspbian Buster Desktop my Rpi4 is producing some strange behaviour:

After it froze on restore, started with a new SD built from master and still shutdown or reboot does not work:

Error during restore:
```
Enabling Services
sudo: unable to resolve host RaspiBlitz: Name or service not known
```

```
$ ssh admin@192.168.1.205
admin@192.168.1.205's password: 
Linux RaspiBlitz 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Nov 27 13:36:39 2019 from 192.168.1.142
Starting the main menu ...
***********************************************************
WARNING: NO HDD FOUND -> Shutdown, connect HDD and restart.
***********************************************************
admin@RaspiBlitz:~ $ sudo shutdown now
admin@RaspiBlitz:~ $ sudo reboot
admin@RaspiBlitz:~ $ 
```

Built from previous commits too and both with the user `pi` and `root`, with similar results.
The problems are not present before running the build_sdcard.sh so I suspect it is something in the raspbian or apt upgrade breaking the setup.

Curious if you can reproduce?

openoms, 2019-11-27:
The worst is that having the same problem with the 1.3 image with this kernel:
Linux RaspiBlitz 4.19.66-v7l+ #1253 SMP Thu Aug 15 12:02:08 BST 2019 armv7l

It did freeze on restore at this:
https://termbin.com/t0q5

I have one of the Shim-fan-s which are producing the flickering.

I have removed plymouth at the first boot still without connecting the SSD.
The flickering has gone, but it does not seem to be the end of it.

I remove the shim fan and the setup continues!

openoms, 2019-11-27:
OK SDcard build is functional from latest master.

The issue was caused by the Shim Fan.

Closing this now.

closed 2019-11-27 by openoms

frennkie, 2019-12-01:
Just tested. Regular build definitely works fine.