raspiblitz/raspiblitz #886: Connect to a Lightning Node failure

smallroy21, 2019-11-27:
trying to connect to the Satoshi's Place node and getting this response

> ******************************
> Connect to A Lightning Node
> ******************************
> 
> COMMAND LINE:
> lncli --chain=bitcoin --network=mainnet connect 024655b768ef40951b20053a5c4b951606d4d86085d51238f2c67c7dec29c792ca@88.98.213.236:9735
> 
> RESULT (might have to wait for timeout):
> [lncli] rpc error: code = Unknown desc = dial tcp 88.98.213.236:9735: connect: connection timed out
> 
> !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
> FAIL
> !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
> Its possible that LND daemon is not running, not configured correct or not connected to the lncli.
> 
> Press ENTER to return to main menu.
> 

Any help appreciated

rootzoll, 2019-11-28:
Activate the RTL Admin GUI under services and use this to manage your channels. The options for this with from RaspiBlitz menus is quite limited and just for basic testing - they dont provide good feebback in case of error.

smallroy21, 2019-11-28:
> Activate the RTL Admin GUI under services and use this to manage your channels. 

When logging into RTL im getting "Error: istrusted: true"

nyxnor, 2021-04-18:
This was proably RTL error. Pls close this.

closed 2021-04-19 by rootzoll