raspiblitz/raspiblitz #887: Enable bitcoind wallet by default

Fonta1n3, 2019-11-28:
Hello,

I have helped a number of your users connect to FullyNoded and its always a bit of a disappointment for them when they realize the bitcoin wallet is disabled. I had a conversation with @rootzoll about this a few months ago and he requested I raise an issue here and that I add the ability to encrypt the wallet.dat file which I have since done. A user can tap "security center" in FullyNoded and add a passphrase, change a passphrase, encrypt and decrypt all using rpc calls to the node via FullyNoded so that the wallet.dat file is being encrypted on the node itself. I understand if you guys do not want to do this and as FullyNoded stands its more designed for power users who understand how a node works.

Anyways just thought I would follow through with our conversation on twitter and see if this would be possible as FullyNoded and raspiblitz are a great match :) Economic nodes are great too. Cheers for the awesome project.

openoms, 2019-11-28:
I think it can be still activate on-demand either when pairing Fully noded or activating Joinmarket. 
We don`t use the bitcoind wallet to anything else for now.

Fonta1n3, 2019-11-28:
Perfect, I have added in the compatibility with disablewallet=1 anyway just incase. Uploading now.

closed 2019-12-09 by rootzoll