raspiblitz/raspiblitz #895: after recover from .gz file - LND error

btc21cluib, 2019-12-04:
Hi
after I recover my lnd wallet from a lnd-rescue-xx.tar.gz file 
I got this lnd service error 
https://termbin.com/yugl
any help?
I reload the HDD - blockchain  already 3 times but without success.

openoms, 2019-12-04:
not sure what this is:
```
Dec 04 10:21:08 21btcclub lnd[19139]:         github.com/lightningnetwork/lnd@/server.go:1180 +0x64
Dec 04 10:21:08 21btcclub lnd[19139]: github.com/lightningnetwork/lnd.Main(0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
Dec 04 10:21:08 21btcclub lnd[19139]:         github.com/lightningnetwork/lnd@/lnd.go:620 +0x15f0
Dec 04 10:21:08 21btcclub lnd[19139]: main.main()
Dec 04 10:21:08 21btcclub lnd[19139]:         github.com/lightningnetwork/lnd@/cmd/lnd/main.go:14 +0x2c
Dec 04 10:21:08 21btcclub systemd[1]: lnd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Dec 04 10:21:08 21btcclub systemd[1]: lnd.service: Failed with result 'exit-code'.
Dec 04 10:22:08 21btcclub systemd[1]: lnd.service: Service RestartSec=1min expired, scheduling restart.
```
but seems to be an issue with the LND installation.
Make sure you are not trying to recover the database in an older version of LND.
You can update LND to the latest release with this:
https://github.com/openoms/lightning-node-management/blob/master/lnd.updates/README.md#update-lnd-to-v081-beta

btc21cluib, 2019-12-09:
Thanks for you help - update  to LND 8.0, then erase LND data and start recover with seed and Channel backup file helps. Now the node is running, but noch channel no funds.
I wait now since 3 days but nothing. 
How long can tis process take, or do I need to fund my wallet again and open a new channel?
thanks for any help 
still wait for recovering my channel funds.

btc21cluib, 2019-12-15:
so I open a new channel and wait. but none of my lost channels with the funds  comes back.
What can I try now?
my wallet 
Public Key: 027be9e7e2bc7ec87177ac298c01517ef46f979e1e3acec46efcf7a6f1b3ddadff
11 open and running Peer 
1 open channel

nyxnor, 2021-04-18:
@btc21cluib Did you get your funds back? Issue still open?

nyxnor, 2021-05-17:
@rootzoll Please close this.

closed 2021-05-17 by rootzoll