raspiblitz/raspiblitz #898: RaspiBlitz slow boot and acts weird after boot

got3nks, 2019-12-04:
I have setup three times this RaspBerry PI 4 (4GB ram + 1 TB SSD + 32 GB Micro-SD) with latest version of RaspiBlitz (raspiblitz-v1.3-2019-09-05.img.gz).

Setup is always fast, but after initialization and reboot it takes very long before I can log in SSH. Even after succesfull boot I sometime gets error like "bitcoind is not running" while some other times I can see main menu (and node is sync'd: **bitcoin v0.18.1 mainnet Sync OK 100.00%**). 

The LCD screen also acts weird. I dont see the boot logo and after long time even if node is running correctly (I can see the status in SSH as said before) this is what I get on the screen.

[image]

I think the culprit is in systemd, maybe some services are not booting up correctly.

```
admin@ViresInNumeris:~ $ sudo systemctl status bitcoind
Failed to get properties: Connection timed out
```

I already tried clearing HDD (except blockchain) and re-flash SD card with RaspiBlitz several times. I always end up having this issue.

﻿﻿XXdebugLogs.sh sometime could check systemd status, sometime could not (without reboot, it seems being an on-and-off issue). When it works this is the output (as you can see all services are actually running fine).

```
***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.3
chainnetwork: bitcoin / main
 14:53:44 up  1:01,  2 users,  load average: 2.00, 2.08, 1.52

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-12-04 14:36:47 GMT; 17min ago
  Process: 1694 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 1700 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 1727 (bitcoind)
    Tasks: 18 (limit: 4915)
   Memory: 3.2G
   CGroup: /system.slice/bitcoind.service
           └─1727 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Dec 04 14:36:47 ViresInNumeris systemd[1]: Starting Bitcoin daemon...
Dec 04 14:36:47 ViresInNumeris systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Wed 2019-12-04 13:52:27 GMT, end at Wed 2019-12-04 14:53:56 GMT. --
Dec 04 14:36:47 ViresInNumeris systemd[1]: Starting Bitcoin daemon...
Dec 04 14:36:47 ViresInNumeris systemd[1]: Started Bitcoin daemon.
1575470634 ERROR: Error found in Logs
1575471065 ERROR: Error found in Logs

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2019-12-04T14:38:14Z UpdateTip: new best=00000000000000000008f9948017a6aa334b7ac1c8e1be8eafce1fd58064b146 height=606618 version=0x20000000 log2_work=91.404403 tx=480690884 date='2019-12-04T14:01:14Z' progress=0.999983 cache=1.7MiB(15137txo) warning='62 of last 100 blocks have unexpected version'
2019-12-04T14:38:20Z UpdateTip: new best=000000000000000000092898b48ab1e9ca45ab0d1edcd9cc0eed9638d498a016 height=606619 version=0x20002000 log2_work=91.404427 tx=480693188 date='2019-12-04T14:03:48Z' progress=0.999984 cache=2.7MiB(23732txo) warning='62 of last 100 blocks have unexpected version'
2019-12-04T14:38:27Z UpdateTip: new best=00000000000000000009adc38ce91a682772ef6359622f654fec3c77e8dfbb6e height=606620 version=0x20000000 log2_work=91.404452 tx=480695421 date='2019-12-04T14:04:03Z' progress=0.999984 cache=3.5MiB(30808txo) warning='61 of last 100 blocks have unexpected version'
2019-12-04T14:38:33Z UpdateTip: new best=0000000000000000000f855b5fe7165f89ba28de0d34207a85c21e22cdbd1254 height=606621 version=0x20400000 log2_work=91.404476 tx=480698284 date='2019-12-04T14:24:47Z' progress=0.999994 cache=4.5MiB(39882txo) warning='62 of last 100 blocks have unexpected version'
2019-12-04T14:38:40Z UpdateTip: new best=00000000000000000014a66e3ca0413f7e46100fc3c697c715c52bf2b51b8cb8 height=606622 version=0x3fffe000 log2_work=91.404501 tx=480700808 date='2019-12-04T14:27:35Z' progress=0.999995 cache=5.5MiB(49097txo) warning='62 of last 100 blocks have unexpected version'
2019-12-04T14:38:40Z New outbound peer connected: version: 70015, blocks=606623, peer=2
2019-12-04T14:38:41Z Pre-allocating up to position 0x2000000 in blk01885.dat
2019-12-04T14:38:49Z Pre-allocating up to position 0x300000 in rev01885.dat
2019-12-04T14:38:49Z UpdateTip: new best=0000000000000000000019083464a4c259f8057f84a55e3ac137b47a6883454e height=606623 version=0x20c00000 log2_work=91.404525 tx=480702774 date='2019-12-04T14:28:29Z' progress=0.999995 cache=6.6MiB(57575txo) warning='63 of last 100 blocks have unexpected version'
2019-12-04T14:38:49Z New outbound peer connected: version: 70015, blocks=606623, peer=4
2019-12-04T14:38:55Z New outbound peer connected: version: 70015, blocks=606623, peer=6
2019-12-04T14:38:56Z New outbound peer connected: version: 70015, blocks=606623, peer=7
2019-12-04T14:39:14Z New outbound peer connected: version: 70015, blocks=606623, peer=8
2019-12-04T14:39:16Z New outbound peer connected: version: 70015, blocks=606623, peer=9
2019-12-04T14:39:22Z New outbound peer connected: version: 70015, blocks=606623, peer=10
2019-12-04T14:42:59Z UpdateTip: new best=00000000000000000002a43bc2d047a50c7d620f931a899c03e813a117a4807b height=606624 version=0x3fffe000 log2_work=91.40455 tx=480705821 date='2019-12-04T14:42:08Z' progress=1.000000 cache=7.6MiB(66645txo) warning='64 of last 100 blocks have unexpected version'
2019-12-04T14:49:02Z UpdateTip: new best=00000000000000000012a0401b977c3ef83116325a815dde38fdd1a8bae21d07 height=606625 version=0x3fffe000 log2_work=91.404575 tx=480708814 date='2019-12-04T14:49:12Z' progress=1.000000 cache=8.4MiB(74777txo) warning='65 of last 100 blocks have unexpected version'
2019-12-04T14:50:46Z UpdateTip: new best=0000000000000000000d81463e0e644f3f9f4a3be9cb4c06c6d41c516d6a5d32 height=606626 version=0x20800000 log2_work=91.404599 tx=480710985 date='2019-12-04T14:50:28Z' progress=1.000000 cache=9.0MiB(80181txo) warning='65 of last 100 blocks have unexpected version'
[... 176 lines cut ...]

*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0
raspberryPi=4
powerFAIL=0
powerWARN=0
powerMIN='875300 microVolt'
tempFAIL=0
tempWARN=0
tempMAX='4700 centiGrad'

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.1.175'
tempCelsius='46.7'
uptime=4656
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=0
bitcoinErrorShort='Error found in Logs'
bitcoinErrorFull='0'
# OK: log 'ERROR: Error found in Logs' written to /home/admin/systemd.blockchain.log
startcountLightning=1
lndActive=0
scriptRuntime=181
```

openoms, 2019-12-04:
Are you using a shim fan? Try removing it! Very much looks like the random issues documented in this thead: https://github.com/rootzoll/raspiblitz/issues/796

For now don`t worry about the RPi getting hot without the fan, it will slow down instead of getting damaged. Use and open case and possibly a USB fan or else. We are looking for the ideal solution.  Also recently ordered another shim-fan to see if it is just (quite common) sporadic issue or a design problem.

got3nks, 2019-12-04:
Yes I'm using a shim fan, indeed.

I checked the thread and it looks like a workout would be to uninstall Plymouth?

openoms, 2019-12-04:
As you can see from my comments https://github.com/rootzoll/raspiblitz/issues/796#issuecomment-560563554 I had further problems after removing plymouth got me through the boot process.
The cause is not plymouth, but likely the shimfan interfering with the LCD on the GPIO.
Not sure if it affects all the new fans, but as far as I can tell only removing/changing can solve the problem completely.

To remove plymouth run:
`sudo apt-get purge plymouth* `
and restart.

got3nks, 2019-12-04:
What issues did you experience after removing plymouth?

openoms, 2019-12-05:
> What issues did you experience after removing plymouth?
> 
> After uninstalling plymouth and installing FanShim drivers I didn't experience any issues so far.

See this comment https://github.com/rootzoll/raspiblitz/issues/796#issuecomment-562146288
and please report if you experience similar.

got3nks, 2019-12-05:
Only shutdown and reboot issues? I have shut down my node from the RaspiBlitz menu succesfully a few times.

got3nks, 2019-12-06:
Well you are right, having issues with shuthdown now.

openoms, 2019-12-06:
Suboptimal but best way to pull the plug is to disconnect the network cable first. Wait for the HDD led to stop blinking. Pull the power only then.

`sudo reboot -f` found to work for restart.

openoms, 2021-11-08:
solved and outdated

closed 2021-11-08 by openoms