raspiblitz/raspiblitz #899: White Screen

justinsane, 2019-12-06:
Hi everyone,

My node used to work but recently I'm getting a blank white screen (see image below). I'm connected to ethernet. Can anyone advise on troubleshooting steps?

I have a feeling my SD card is messed up? 

As I mentioned, I used to SSH in remotely and everything worked fine. Not sure what happened.  Thanks in advance.

[image]

rootzoll, 2019-12-09:
This is what it looks when the RaspiBlitz shutdown (and is waiting for you to disconnect and reconnect power). Or when during startup the system does not reach the point where its loading the the software drivers. Are you still able to SSH in?

justinsane, 2019-12-10:
I'm not able to SSH in. 

I checked the devices connected to my router and the Raspberry Pi wasn't one of them (while connected via Ethernet). 

Is my SD or HDD corrupt since it won't boot? Any idea on the next logical troubleshooting step? 

Thanks in advance!

rootzoll, 2019-12-10:
If the screen is white and you cant SSH in it looks like the system is not able to boot up and first thing to check/re-write/replace would be the sd card in that case.