raspiblitz/raspiblitz #901: sd card build script: Remove of old Swiss LCD option

rootzoll, 2019-12-10:
The play-zone.ch LCD is no longer part of the swiss shopping list - so remove the different driver option from the sd build script to make it more simple and remove dependencies. Also unify the LCD driver install for raspbian and dietpi base image to reduce complexity.

rootzoll, 2019-12-10:
@frennkie i committed the updated sd build script for this issue in the v1.4 branch. I refactored the dietpi part also. I will test on normal raspbian image. It would be great if you can check for dietpi if possible - thx.

phenomeus, 2020-01-17:
I would like to help testing it on a clean dietpi 6.28.0 on a rpi4 4GB

rootzoll, 2020-01-20:
@phenomeus get in sync with @openoms - he keeps track of the dev on other platforms like dietpi

openoms, 2020-01-20:
DietPi for the RPi is based on Raspbian and cleaned with a script (similar to our build_sdcard.sh).
I am not planning to test beside Raspbian as that is getting the fixes and updates first and I would not advise to use the extra software of DietPi as it can break the RaspiBlitz and open new security risks.
Would consider to phase it out too if compatibilty is not maintained.

@phenomeus have a look at this if you like to test anyway: https://github.com/rootzoll/raspiblitz/tree/master/alternative.platforms/dietpi . Would recommend opening a new issue for DietPi related problems (if there is any) to have continuity.

For other SBC-s I found Armbian the most supported and security focused base image, so in alternatives I'd like to maintain that.

phenomeus, 2020-01-26:
while removing old Swiss stuff, it would be nice to add a general question during setup if there is a lcd screen presented at all so the whole process can be skipped/ cleaner installation. I run a pi4 without an always on screen

rootzoll, 2020-01-28:
@phenomeus the v.14 has this new feature that allows to switch between LCD-hat & HDMI output https://github.com/rootzoll/raspiblitz/issues/767

So if you dont have an LCD make sure to switch to HDMI. The best option todo so is to set the `hdmi` file flag on the boot partition of the sd card before booting it up in the RaspberryPi. For details see that feature issue above.

Closing issue for v1.4 release.

closed 2020-01-28 by rootzoll