raspiblitz/raspiblitz #902: Blitz Rescue Script: Import/Export/Backup

rootzoll, 2019-12-10:
Like the `lnd.rescue.sh` script is allowing an export of the LND data to be exported and then imported to a fresh RaspiBlitz. The `blitz.rescue.sh` script allows for a complete export of all RaspiBlitz config and data to then to be imported to a fresh RaspiBlitz.

rootzoll, 2019-12-19:
Worked while testing a manual migration from Ext4 to BTRFS ... will keep it as a CLI tool for v1.4 release. Might get more integrated in GUI in future releases.

closed 2019-12-19 by rootzoll

Emzy, 2020-02-13:
You can migrate from ext4 to btrfs in place. And even can rollback to ext4.
No need to delete the data.

https://www.thegeekdiary.com/how-to-convert-ext-file-systems-to-btrfs/

rootzoll, 2020-02-16:
EXPORTING A MIGRATION FILE:
Under REPAIR in the main menu you can since v1.4 choose to download a migration file to your laptop. Then the Blitz powers down. Then you can rebuild/echange your hardware/HDD/SSD.

IMPORTING A MIGRATION FILE:
Make a fresh sd card image and when setting up your new/fresh Blitz since v.4 it offers (after the power test) to setup the Blitz with a "MIGRATION" file. The user can then choose to format the HDD with EXT4 or BTRFS and upload the migration file. Then you choose how to get the blockchain data again. Choose COPY to copy from your laptop or another blitz over LAN (about 4 hours) or choose SYNC and let the RaspiBlitz sync and validate the blockchain itself (just recommended on RaspiBlitz4+SSD and will still take around 2-3 days).