raspiblitz/raspiblitz #905: RPi4 build with SATA-USB expansion board and 5V fan

openoms, 2019-12-11:
In connection with the metal case (https://diynodes.com/product/zkds-metal-case-for-the-raspiblitz/) I have been experimenting with this SATA-USB expansion board and found it very stable.

Sources:
Amazon:
https://www.amazon.co.uk/Geekworm-Raspberry-Expansion-USB3-0-Shield/dp/B07VXF2HJG/
DIYnodes.com (bitcoin only): https://diynodes.com/product/sata-usb-expansion-board-and-usb-bridge/

It has 2 pairs of 5V power pins too, so there are more options to attach a fan.
This one seems to be good quality, silent and efficient (keeps the temp at 58-62 degrees max with repeated stresstest runs `./config.scripts/blitz.stresstest.sh`):
https://www.ebay.co.uk/itm/Square-Cooling-Fan-with-Heatsink-Kit-For-Raspberry-Pi-4B-PEONCHU-lq/123960611223
It sits on a heatsink which has a sticky layer on the bottom.
there can be plenty of similar fans be found on amazon/aliexpress. 
[image]
[image]

I am also using some 22mm x M2.5 screws (with M2.5 locking nuts) to support the screen from below. This keeps it very stable when pressed.

openoms, 2020-02-03:
More for the Maximal package. BIg SSd-s seem to draw siginificantly more power than the smaller ones.
500GB SSD-s are taking 1-1.2 Amperes.
1TB I have seen are 1.6-1.7 (and 1.7 was causing problems)

I have tested 3 types of 1 TB SSD-s with this setup so far:

**GOOD**:
SanDisk SSD PLUS 1 TB Sata III 2.5 Inch Internal SSD, Up to 535 MB/s: https://www.amazon.co.uk/SanDisk-SDSSDA-1T00-G26-Internal-Speed-Write/dp/B07D998212
power req: **1.6 A** - no issues so far

WD Blue 3D NAND Internal SSD 2.5 Inch SATA - 1 TB
https://www.amazon.co.uk/gp/product/B073SBQMCX/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1
power req: **1.6 A** - no issues so far
[image]

**BAD**:
Crucial MX500 CT1000MX500SSD1 1 TB (3D NAND, SATA, 2.5 Inch, Internal SSD)
https://www.amazon.co.uk/Crucial-MX500-CT1000MX500SSD1-NAND-Internal/dp/B078211KBB
power req: **1.7 A** - on high load of the node (electrs + txindex indexing at the same time)  had read errors and froze (probably due to the power spike)
[image]

The power was a single 3A Pimoroni PSU from the shopping list for all. The X825 board can be powered separately too with a 5V barrel connector in case that would be needed in the future. Already very useful when cloning the disks.

openoms, 2020-04-10:
Link to a build with the X825 board, heatsink-case and 22mm * M2.5 screws working together well:
https://twitter.com/DIYnodes/status/1248587471983607808

Fits the metal case too.

openoms, 2020-09-30:
note from TG regarding the Geekworm Raspberry Pi 4 X872 M.2 NVMe SSD Expansion Board https://www.amazon.co.uk/Geekworm-Raspberry-Expansion-Compatible-Support/dp/B0855DMH6C

https://t.me/raspiblitz/40612
[image]

marxenegls, 2021-09-14:
i had a similar issue with fan power inputs being used by the tft screen - in the end i stripped an old usb cable to power the internal fan i have a second external fan fixed in place temp never exceeds 47*