raspiblitz/raspiblitz #908: Channel.db appears corrupt after upgrade to v1.2

sicXnull, 2019-12-15:
Upgraded to v1.2 and cannot start the wallet with my channel.db from v1.1

Renaming channel.db to channel.db.old fires up just fine, obviously without channels and related funds. The error appears to be 

`Dec 15 20:24:49 RaspiBlitz lnd[28919]: panic: page 12409 already freed`

Any way to recover? full log below. 

```
***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.2
chainnetwork: bitcoin / main
 20:25:06 up 10 min,  2 users,  load average: 3.67, 3.09, 1.66

*** BCLOCKHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-12-15 20:16:41 GMT; 8min ago
  Process: 4984 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
  Process: 4977 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
 Main PID: 5048 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─5048 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Dec 15 20:16:40 RaspiBlitz systemd[1]: Starting Bitcoin daemon...
Dec 15 20:16:41 RaspiBlitz systemd[1]: Started Bitcoin daemon.

*** LAST BCLOCKHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Thu 2016-11-03 17:16:43 GMT, end at Sun 2019-12-15 20:25:06 GMT. --
Dec 15 20:16:40 RaspiBlitz systemd[1]: Starting Bitcoin daemon...
Dec 15 20:16:41 RaspiBlitz systemd[1]: Started Bitcoin daemon.

*** LAST BCLOCKHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2019-12-15T20:21:07Z net thread start
2019-12-15T20:21:07Z opencon thread start
2019-12-15T20:21:07Z dnsseed thread start
2019-12-15T20:21:07Z msghand thread start
2019-12-15T20:21:07Z addcon thread start
2019-12-15T20:21:07Z init message: Done loading
2019-12-15T20:21:10Z New outbound peer connected: version: 70015, blocks=608255, peer=0
2019-12-15T20:21:11Z txindex is enabled at height 608255
2019-12-15T20:21:11Z txindex thread exit
2019-12-15T20:21:18Z Loading addresses from DNS seeds (could take a while)
2019-12-15T20:21:18Z 242 addresses found from DNS seeds
2019-12-15T20:21:18Z dnsseed thread exit
2019-12-15T20:21:22Z New outbound peer connected: version: 70015, blocks=608255, peer=2
2019-12-15T20:21:35Z New outbound peer connected: version: 70015, blocks=608255, peer=4
2019-12-15T20:21:49Z New outbound peer connected: version: 70015, blocks=608255, peer=5
2019-12-15T20:21:51Z New outbound peer connected: version: 70015, blocks=608255, peer=6
2019-12-15T20:21:55Z New outbound peer connected: version: 70015, blocks=608255, peer=7
2019-12-15T20:22:53Z UpdateTip: new best=0000000000000000000200dd6b95faf024f3d611c500969a1485ccac02df6121 height=608256 version=0x3fffc000 log2_work=91.44378 tx=484122727 date='2019-12-15T20:21:25Z' progress=1.000000 cache=1.3MiB(11974txo) warning='57 of last 100 blocks have unexpected version'
2019-12-15T20:22:54Z New outbound peer connected: version: 70015, blocks=608256, peer=8
2019-12-15T20:22:54Z New outbound peer connected: version: 70015, blocks=608256, peer=9

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2019-12-15 20:24:49 GMT; 17s ago
  Process: 28919 ExecStart=/usr/local/bin/lnd --externalip=${publicIP}:${lndPort} (code=exited, status=2)
  Process: 28915 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 28919 (code=exited, status=2)

Dec 15 20:24:49 RaspiBlitz systemd[1]: lnd.service: Unit entered failed state.
Dec 15 20:24:49 RaspiBlitz systemd[1]: lnd.service: Failed with result 'exit-code'.

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Thu 2016-11-03 17:16:43 GMT, end at Sun 2019-12-15 20:25:07 GMT. --
Dec 15 20:24:49 RaspiBlitz systemd[1]: Started LND Lightning Daemon.
Dec 15 20:24:49 RaspiBlitz lnd[28919]: panic: page 12409 already freed
Dec 15 20:24:49 RaspiBlitz lnd[28919]: goroutine 1 [running]:
Dec 15 20:24:49 RaspiBlitz lnd[28919]: github.com/coreos/bbolt.(*freelist).free(0x35c7940, 0xd7508b, 0x0, 0x61c6c000)
Dec 15 20:24:49 RaspiBlitz lnd[28919]:         /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v1.3.2/freelist.go:175 +0x448
Dec 15 20:24:49 RaspiBlitz lnd[28919]: github.com/coreos/bbolt.(*node).spill(0x35c7a00, 0x3621b8c, 0x8)
Dec 15 20:24:49 RaspiBlitz lnd[28919]:         /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v1.3.2/node.go:363 +0x204
Dec 15 20:24:49 RaspiBlitz lnd[28919]: github.com/coreos/bbolt.(*Bucket).spill(0x387418c, 0xbf75c384, 0x10d16ae2)
Dec 15 20:24:49 RaspiBlitz lnd[28919]:         /Users/roasbeef/gocode/pkg/mod/github.com/coreos/bbolt@v1.3.2/bucket.go:568 +0x360
Dec 15 20:24:49 RaspiBlitz systemd[1]: lnd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Dec 15 20:24:49 RaspiBlitz systemd[1]: lnd.service: Unit entered failed state.
Dec 15 20:24:49 RaspiBlitz systemd[1]: lnd.service: Failed with result 'exit-code'.

*** LAST 30 LND INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
2019-12-15 20:21:47.812 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-12-15 20:21:47.885 [INF] CHDB: Checking for schema update: latest_version=8, db_version=7
2019-12-15 20:21:47.886 [INF] CHDB: Performing database schema migration
2019-12-15 20:21:47.886 [INF] CHDB: Applying migration #8
2019-12-15 20:21:47.886 [INF] CHDB: Migrating to the gossip message store new key format
2019-12-15 20:21:47.886 [INF] CHDB: Migration to the gossip message store new key format complete!
2019-12-15 20:21:47.890 [INF] LTND: Shutdown complete
2019-12-15 20:22:48.418 [INF] LTND: Version: 0.6.0-beta commit=v0.6-beta, build=production, logging=default
2019-12-15 20:22:48.418 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-12-15 20:22:48.423 [INF] CHDB: Checking for schema update: latest_version=8, db_version=7
2019-12-15 20:22:48.423 [INF] CHDB: Performing database schema migration
2019-12-15 20:22:48.425 [INF] CHDB: Applying migration #8
2019-12-15 20:22:48.425 [INF] CHDB: Migrating to the gossip message store new key format
2019-12-15 20:22:48.425 [INF] CHDB: Migration to the gossip message store new key format complete!
2019-12-15 20:22:48.428 [INF] LTND: Shutdown complete2019-12-15 20:23:48.813 [INF] LTND: Version: 0.6.0-beta commit=v0.6-beta, build=production, logging=default
2019-12-15 20:23:48.814 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-12-15 20:23:48.818 [INF] CHDB: Checking for schema update: latest_version=8, db_version=7
2019-12-15 20:23:48.818 [INF] CHDB: Performing database schema migration
2019-12-15 20:23:48.818 [INF] CHDB: Applying migration #8
2019-12-15 20:23:48.819 [INF] CHDB: Migrating to the gossip message store new key format
2019-12-15 20:23:48.819 [INF] CHDB: Migration to the gossip message store new key format complete!
2019-12-15 20:23:48.824 [INF] LTND: Shutdown complete
2019-12-15 20:24:49.312 [INF] LTND: Version: 0.6.0-beta commit=v0.6-beta, build=production, logging=default
2019-12-15 20:24:49.312 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-12-15 20:24:49.370 [INF] CHDB: Checking for schema update: latest_version=8, db_version=7
2019-12-15 20:24:49.370 [INF] CHDB: Performing database schema migration
2019-12-15 20:24:49.370 [INF] CHDB: Applying migration #8
2019-12-15 20:24:49.370 [INF] CHDB: Migrating to the gossip message store new key format
2019-12-15 20:24:49.370 [INF] CHDB: Migration to the gossip message store new key format complete!
2019-12-15 20:24:49.374 [INF] LTND: Shutdown complete
*** LAST 20 RTL LOGS ***
-- Logs begin at Thu 2016-11-03 17:16:43 GMT, end at Sun 2019-12-15 20:25:07 GMT. --
Dec 15 20:16:41 RaspiBlitz systemd[1]: Started RTL daemon.

*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0
powerFAIL=0
powerWARN=2
powerMIN='1200000 microVolt'
tempFAIL=7
tempWARN=14
tempMAX='7410 centiGrad'
IMPORTANT: There are some hardware issues with your setup.
'Run Hardwaretest' in main menu or: sudo /home/admin/05hardwareTest.sh

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='deleted'
tempCelsius='68.7'
uptime=638
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=608256
initialSync=0
syncProgress=100.00
startcountLightning=9
lndActive=0
scriptRuntime=2

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999
```

openoms, 2019-12-16:
If your channel db is corrupted then it won`t be possible to restore with any version.

The relevant information here is the LND version.
RaspiBlitz 1.2 has lnd 0.6.0, RaspiBlitz 1.1 had 0.5.1 I think.

It is unclear what is causing the lnd error from this log collection.
Waht can be seen that the database migration has completed successfully on the lnd upgrade and according to that channel.db does not appear to be corrupt.

Did you upgrade lnd separately on 1.1 maybe?

sicXnull, 2019-12-16:
Okay, thank you. If #910 is correct then this issue will solve itself.

rootzoll, 2019-12-16:
When RaspiBlitz is doing an update (fresh sd card) it should create a "backup_lnd" folder on your HDD. Check if you find in there an old version of the channel.db ... normally you should not use old versions, but in your case it might be a chance.

sicXnull, 2019-12-17:
Yes, i made multiple backups of each folder before doing an install, which is the weird part.

closed 2019-12-22 by sicXnull