raspiblitz/raspiblitz #909: Test USB3 performance of HDD/SSD

rootzoll, 2019-12-15:
Make sure that the HDD/SSD is connected performant and not on one of the USB2 ports .. see:
https://stadicus.github.io/RaspiBolt/raspibolt_20_pi.html#make-sure-usb3-is-performant

rootzoll, 2020-04-01:
Marked this idea for the Lightning HackSprint - details see: https://wiki.fulmo.org/index.php?title=Lightning_HackSprint

The tasks would be: Add a short test if the HDD/SSD is connected on the USB3 port (on RP4) to the `/home/admin/config.scripts/blitz.stresstest.sh` and add the result with warnings if needed to the `home.admin/05hardwareTest.sh`

rootzoll, 2020-07-21:
Proof that this is a real life issue: https://twitter.com/odudex/status/1285554970251743232?s=20

frennkie, 2020-09-19:
Another one 

ngl, it'd been several tough but exciting weeks learning the lightning ways; finally managed to tame this bad boy. Cheers to @RaspiBlitz @ln_zap @openoms for the invaluable efforts and insights. First payment at @bitcoinemb went silk-smooth and truly lightning fast!⚡ https://t.co/5UMYxwwKcR

https://twitter.com/moizen_ssk/status/1307164119229841408

It'll try to look into this over the weekend.

openoms, 2020-09-19:
@frennkie that adapter would not work on USB3 so good job on him trying the USB2.
Slower is better than errors all over the place.

frennkie, 2020-09-19:
Wow... How can you tell? 

Anyway... A speed rest would still show that he has a ext. disk that performa way worse taht it's potential.

So we should recommend that he gets a USB3 case, or no? (and use a different port)

openoms, 2020-09-19:
I have tested a couple of these :)
https://twitter.com/openoms/status/1158876381306064897?s=19

frennkie, 2020-09-19:
I created a [script](https://gist.github.com/frennkie/57698603f59efa72606192d2a172025c) that parses the output of `/usr/bin/usb-devices`. Looks like this:

```
$ ./parse-usb-devices.sh
Port1-2-2 USB@480Mbit/s: Silicon Motion, Inc. - Taiwan (formerly Feiya Technology Corp.) Flash Drive  (Serial: XXX)
Port2-1-1 USB@5000Mbit/s: Samsung Electronics Co., Ltd Portable SSD T5  (Serial: XXX)
```

When I moved the additional USB I connected for testing to the USB3 port it changes to this:

```
$ ./parse-usb-devices.sh
Port2-1-0 USB@5000Mbit/s: Silicon Motion, Inc. - Taiwan (formerly Feiya Technology Corp.) Flash Drive  (Serial: XXX)
Port2-1-1 USB@5000Mbit/s: Samsung Electronics Co., Ltd Portable SSD T5  (Serial: XXX)
```

@rootzoll How and where can we include this? Does it still make sense to run a performance benchmark (e.g. `hdparm`)?

rootzoll, 2020-09-20:
> https://www.jeffgeerling.com/blog/2020/raspberry-pi-usb-boot-uasp-trim-and-performance

It could be part of `blitz.stresstest.sh` or gets at least also before the setup like the stresstest: Letting the user know that the HDD/SSD is connected to the wrong/slow USB port. Then it could say "please change ports" and offer `ignore` and `restest` .