raspiblitz/raspiblitz #910: Missing funds & New Transactions after upgrade

sicXnull, 2019-12-16:
Upgraded to v1.3, had problems so downgraded to V1.2

Synced up, zero funds and now noticed 3 different TX's. The odd part is, the TX's are right about the same times that i booted up v1.3 (two last night), then booted up v1.2 (today)

I've been trying to make sense of this, but can't at the moment. To me, it looks like the funds were stolen, but the timing is way too close to the times i made installation. 

Am i missing something here?

Note: The img downloads were from the releases page, direct download, no torrent.

[image]

[image]

I also tried recovering funds from seed on both the pi and lnd wallet desktop. Shows the same TX's

Here are the transaction is you want to dig in

Funding

```
Dest Addresses
:
35JaP8txyaYmmQQ2uCJbfHsSBhuDuniG5j
bc1qyf4qvq9l4ghxyrm60rcrmggh8nphtfkmaxxnxt 
Time Stamp
:
Apr/27/19 01:52:45 
Num Confirmations
:
34,854
Block Hash
:
00000000000000000020d07dc30765aff541b320d91fe878383972c6caf6039a 
Block Height
:
573,433
Tx Hash
:
19aef94b4a101ba2c73f806e968c0d5e633e0aa58e8e92b4e99a26757067d4fe 
Amount
:
353,039

```

Transactions yesterday

```
Dest Addresses
:
bc1qcuman5kplk6epk9h57z58xpw9aeuavqxyl0durjck8fztwhznxeqhjm4aw
bc1qydlufmzfcfmrweaqyhnfeuva0wm6yqapf76pn4 
Time Stamp
:
Dec/14/19 12:08:42 
Num Confirmations
:
165
Total Fees
:
2,866
Block Hash
:
00000000000000000003d058cd9ab54c7711f00739ed8f29cedebdef84ce24d8 
Block Height
:
608,122
Tx Hash
:
00aba01cc0d5d1b695ad0495ccbfa82b957fccb392ff7a207e8f8ca53d287c0c 
Amount
:
-48333 
```
```
Dest Addresses
:
bc1q7na6kzwj73x52dg88n90jfkyt0vsm8jux2mwll
bc1qn03fnexjrsc0maxa0f5trwzxnvjyu6p9wwlwy23fd5qml9u3mqjq9k4q7u 
Time Stamp
:
Dec/14/19 12:08:42 
Num Confirmations
:
165
Total Fees
:
1,981
Block Hash
:
00000000000000000003d058cd9ab54c7711f00739ed8f29cedebdef84ce24d8 
Block Height
:
608,122
Tx Hash
:
a14e529fbdc93728f9dea8023bdf0898097a10e8586a8c420bcd6a2bfad69bc0 
Amount
:
-77759 
```

Transaction today

```
INFO
close
Dest Addresses
:
bc1qny392lp5hcqzsp4rdag0xdzt53p9cuduwxmkxh
bc1qr76a62zpaqx9lch3l3442redsxs9x432e9f3ye8r6pt04rktmxtqk6sa5g 
Time Stamp
:
Dec/15/19 14:44:31 
Num Confirmations
:
33
Total Fees
:
2,932
Block Hash
:
000000000000000000072b1305588ccf14d99e9143298a2a7a85e61edaf01328 
Block Height
:
608,254
Tx Hash
:
2bf93840a10d9b7f9eb4630633dd9cb41227196bd30d06f69c2b0fdf22d1c3b4 
Amount
:
-100206 ```

openoms, 2019-12-16:
Your channels have been force closed from the Static Channel Backups and funds will become available in the onchain part of you LND wallet after the locktime expires. 

It is a day by deafult, so funds can be already there, but can be longer.

Check again.

sicXnull, 2019-12-16:
Oh. That's awesome. I was hoping that was the case. Thanks for the help!

sicXnull, 2019-12-22:
Funds still have not returned, but this may be from the pending channels. Closing for now. Thanks for the help

closed 2019-12-22 by sicXnull