raspiblitz/raspiblitz #922: Upgrade rpi3 to rpi4 and HDD to SSD - How can I move my node to the new hardware?

DoktorWhat, 2019-12-20:
Is it possible to move a node from old to new hardware without having to close all channels and losing all inbound liquidity in the process? I am very hesitant about trying to migrate the software to a new instance without asking here first.

currently running on Raspiblitz v1.3 (rpi3, HDD)

btc21cluib, 2019-12-21:
since I upgrade with ssd and the recommended case my node is not working anymore debug: /home/admin/XXdebugLogs.sh | nc termbin.com 9999

rootzoll, 2019-12-22:
At the moment you just have the `lnd.rescue.sh` script that can be called under `MAIN MENU > REPAIR > BACKUP` - it creates a zip-tar file of your LND directory that you can download to your laptop. So then you change your hardware after that (rpi3 -> rpi4 or HDD -> SSD) start with a fresh SD card image and during setup when asked about LND you say `old wallet` and then upload the LND rescue file to your new setup. This is part of the RaspiBlitz v1.3 ... but it only moves your LND data (funds and channels) - not anything else of your node setup.

With the next RaspiBlitz version 1.4 there will also be the `blitz.rescue.sh` script. It will backup all of the RaspiBlitz data and help to move it to a new node.

Lupus-Pluvia, 2020-01-28:
hello wanted to ask where the backup is saved when you download it can't find it

rootzoll, 2020-01-31:
When you start downloading the backup with the scp command from the terminal it gets downloaded to the directory you are in at that moment. So if you want the download in a special folder, best is first to navigate into that folder in your terminal and then start the download. If you havent changed any directory on your laptop terminal before download, its most probably in your user directory.

closed 2020-01-31 by rootzoll