raspiblitz/raspiblitz #924: Flash memory reliability - MLC USB drive recommendations

openoms, 2019-12-22:
Aiming to collect some of the more reliable flash drive options here.

Flash NAND chips (flash memory) seem to have 3 basic groups on the basis of how many bits they store per cell.
Single Level Cell (SLC) are basicly the common SSD-s which are more reliable and fast, but need more power and also more expensive than a USB flash drive.

There is a big difference between Multi Level Cell (quality pendrives) and Triple Level Cell drives (cheapest, but least reliable)

Rewrite cycles are respectively around:
80-100k for SLC
8-10k for MLC
3-5k for TLC (not suitable for constant rewrites even for months)

We should try to indentify some known MLC USB drives for the RAID setup otherwise there will be write errors encountered often.

Good summary here:
https://www.mydigitaldiscount.com/everything-you-need-to-know-about-slc-mlc-and-tlc-nand-flash.html

openoms, 2019-12-23:
SecretKey 32GB MLC USB 21.9 GBP https://www.amazon.co.uk/Drive%EF%BC%8CSecretKey-High-Speed-Storage-Pictures-Security/dp/B07Q1BP91Q

Mushkin impact 64GB MLC USB 23.29 USD https://www.amazon.co.uk/Mushkin-MKNUFDIM64GB-Impact-3-0-Black-Player/dp/B01A69S786

rootzoll, 2019-12-24:
This is good research to have once we make the Raid USB thumb drive part of the official shopping list - targeted for v1.5 release.

rootzoll, 2020-03-16:
BTFS Raid will still be experimental in v1.5 ... moving issue to v1.6

rootzoll, 2020-05-27:
BTFS Raid will still be experimental in v1.6 ... moving issue to Future Release

openoms, 2020-10-24:
Took some measurements and found that even the USB 2.0 32 GB pendrive had power spikes to 0.5 Amper when writing. 
The Sandisk Extreme Pro USB 128 GB SSD did not reach 0.5 Amper, but has a higher sustained power requirement ~ 0.25-0.3 Amper.
A 2 TB 2.5" Toshiba HDD was using 0.5-0.6  Amper during load (labeled to take need up to 1.0A).

For now the Sandisk Extreme Pro USB 128 GB SSD runs stable in a bausy node with many services active. Attached to this build: https://github.com/rootzoll/raspiblitz/issues/905#issue-536472177 and powered only through the USBC (3A PSU).