raspiblitz/raspiblitz #926: Restored wallet from seed and passphrase, can't see any on chain funds

tryolly, 2019-12-24:
Hi, I had to completely redo my raspiblitz and could only restore from the seed + passphrase. I managed to restore from this correctly but it doesn't detect my BIP84 bc1 addresses (located m/84'/0'/0'/0/2 & m/84'/0'/0'/1/2) , I had zero funds on the BIP49 addresses at time of redoing my raspiblitz.

When I go to add funds to the raspiblitz on chain wallet it gives me the first BIP49 address m/49'/0'/0'/0/0 for the 24 word seed + passphrase. Which is great because it means the seed + passphrase worked, but I have already used this address in the past, I thought it should have given me the first unused address? so I am guessing the raspiblitz (and the way I restored it) can't detect any used addresses, or addresses with funds.

Is there a way to get the raspiblitz to detect these addresses?

rootzoll, 2019-12-24:
This looks like something to address with the LND team - do you already tried with the latest LND release (0.8.2). I dont think from the RaspiBlitz side there is much to prevent this. But it would be great to have that fixed. If you can reproduce this behavior on the latest 0.8.2 LND I can try to connect you to a dev from lightning labs.

tryolly, 2019-12-24:
Only tried with LND version 0.7.1, which is the one on my raspiblitz, can I update the version on the raspiblitz to 0.8.2?

I guess I can always sweep the addresses into another wallet. Does anyone know a mobile or online wallet that can sweep bc1 addresses?

rootzoll, 2019-12-26:
You can try to use the new v1.4 Release Candidate image: https://raspiblitz.com/raspiblitz-v1.4RC1-2019-12-25.img.gz ... it works with the latest LND.

tryolly, 2019-12-26:
Cool thanks, I'll give it a crack and then get back to you. Thanks for the help.

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-18 by tryolly