raspiblitz/raspiblitz #936: Bigger default SDcard filesystem needed - 8GB filled up restore

openoms, 2019-12-26:
After installing al services with NodeJS, Rust and .NET the default 8GB partition on the SDcard filled up.

 ```
admin@raspberrypi:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       7.1G  6.8G     0 100% /
devtmpfs        1.9G     0  1.9G   0% /dev
tmpfs           2.0G     0  2.0G   0% /dev/shm
tmpfs           2.0G  217M  1.7G  12% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           2.0G     0  2.0G   0% /sys/fs/cgroup
/dev/mmcblk0p1  253M   52M  201M  21% /boot
/dev/sda1       916G  360G  510G  42% /mnt/hdd
tmpfs           391M     0  391M   0% /run/user/1000
tmpfs           391M     0  391M   0% /run/user/1001
```

```
admin@raspberrypi:~ $ sudo du -h -d1 /
13M     /bin
2.8G    /usr
5.3M    /etc
360G    /mnt
du: cannot access '/proc/1278/task/1278/fd/3': No such file or directory
du: cannot access '/proc/1278/task/1278/fdinfo/3': No such file or directory
du: cannot access '/proc/1278/fd/4': No such file or directory
du: cannot access '/proc/1278/fdinfo/4': No such file or directory
du: cannot access '/proc/2461/task/3023/fdinfo/200': No such file or directory
0       /proc
41M     /opt
340M    /lib
4.0K    /media
3.4G    /home
52M     /boot
4.0K    /C:\RTL\backup
217M    /run
4.0K    /srv
16K     /lost+found
9.5M    /sbin
0       /sys
0       /dev
230M    /var
1.3M    /root
100K    /tmp
367G    /
```

Will either need to run `sudo raspi-config --expand-rootfs`  and restart before the  bootstrap-provision process or build the image on a 16GB SD.

The recommendation (and sensible thing) is using a 32 GB card with the Blitz, so 16 GB image should fit every card.

rootzoll, 2020-01-06:
Yes a restart during the bootstrap-provision or setup with `sudo raspi-config --expand-rootfs` should be done. This is an important TODO before a next RC.

frennkie, 2020-01-06:
In a "standard" setup running `sudo raspi-config --expand-rootfs` should really do no harm..

Maybe just run it and store on the SD card, that it has been run on it. And then the provision script can check this value.

rootzoll, 2020-01-07:
Will add `sudo raspi-config --expand-rootfs` again during normal setup and recovering. Recovering/Update process will check if root fs is below 8GB and reboot in the beginning to make expand before all apps get installed. I will also add a "fsexpanded" flag to the raspiblitz.info (that survives a reboot). So that if the user is using a 8GB card and even after running the expand reboot it still below 8GB that it then outputs a warning: sd card to small.

rootzoll, 2020-01-07:
@openoms the "sudo raspi-config --expand-rootfs" will just work for Raspbian .. what will we do with other systems?

frennkie, 2020-01-07:
I looked into the `raspi-config` script.. It doesn't look too hard to extract it (basically it does some test and then create a script that is run once in the next boot and then deletes itself.

openoms, 2020-01-07:
> @openoms the "sudo raspi-config --expand-rootfs" will just work for Raspbian .. what will we do with other systems?

For the v1.4 SDcard release for the RPi using raspi-config does the job, will need to look into doing it manually fro armbian as there is no similar functionality built-in to `armbian-config`.

rootzoll, 2020-01-08:
tested and works so far good fro me

closed 2020-01-08 by rootzoll

dikkechill, 2020-02-04:
The code isn't working 100% correctly yet. I tested an upgrade with a 8GB sdcard from 1.3 to 1.4 using https://raspiblitz.com/raspiblitz-v1.4RC3-2020-01-29.img.gz.
This resulted in a reboot loop.

The error occurs on line 39 in _bootstrap.provision.sh:   if [ ${fsexpanded} -eq 1 ]; then
The problem is that ${fsexpanded} is never set before it is evaluated.
With an =< 8GB card it then always goes to the else below, trying and failing to expand it before each reboot.

rootzoll, 2020-02-09:
@dikkechill I sd card with min 16GB is recommended

openoms, 2020-05-14:
The command to expand the SDcard filesystem in Armbian is:
`sudo /usr/lib/armbian/armbian-resize-filesystem start`