raspiblitz/raspiblitz #941: 1577634535 ERROR: LND RPC is not responding.

Smiggel, 2019-12-29:
I just setup a new node with a better power supply. I downloaded the blockchain using the torrent option and now the rest is syncing. In the meantime, I watched the status and suddenly got an error screen:

> ****************************************
* The LND service is running with error.
****************************************
If you just started some config/setup, this might be OK.

2 RESTARTS DETECTED - LND might be in a error loop
1577634535 ERROR: LND RPC is not responding. LND may have problems starting up. Check logs, config files and systemd service. Org-Error: [lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"
Dec 29 16:33:07 Treehouse lnd[6121]: Unable to determine if wallet is synced: Post http://127.0.0.1:8332: dial tcp 127.0.0.1:8332: connect: connection refused
err='lnd.conf for user admin is old'
More Error Detail:
err='lnd.conf for user admin is old'

-> Use following command to debug: /home/admin/XXdebugLogs.sh
-> To force Main Menu run: /home/admin/00mainMenu.sh
-> To try restart: sudo shutdown -r now

I uploaded the LND logs:
https://termbin.com/c3k0
https://termbin.com/l2aq

Also here is the debug log:
https://termbin.com/nxpi

When I watch the bitcoin log, I see it's still syncing.

Smiggel, 2019-12-30:
Ok, rebooted my device. Now LND look stable and is running and waiting for the chain sync. Not sure what caused the problem in the first place.

openoms, 2020-01-02:
Come back to this when your blockchain is fully synced. 
Make sure the RPC password is the same in` /mnt/hdd/lnd/lnd.con`f as in `/mnt/hdd/bitcoin/bitcoin.conf`

you can reinitiate the lnd wallet from the menu if you type:
`repair`

closed 2020-05-29 by Smiggel