raspiblitz/raspiblitz #945: ummm not my ip

ConanWolf, 2019-12-31:
in my info i have this...
[image]
where the ip is at... thats not my ip, my ip is 73... 
can i ask whats going on here?

PatrickScheich, 2019-12-31:
Some ISPs do not provide a real (internet routable) IPv4. 

They provide a kind of "private" IPv4 to You/your router (this is the 73... IPv4 address, which you see in your router configuration) and forward all your packets through their gateway (which has the 69.252.68.100) and "all the internet" just sees that IPv4 address.

Probably your ISP also provides you a "real" IPv6 (range) but it is a bit of a hussel to configure this.

ConanWolf, 2019-12-31:
> Some ISPs do not provide a real (internet routable) IPv4.
>
> They provide a kind of "private" IPv4 to You/your router (this is the 73... IPv4 address, which you see in your router configuration) and [...]
so how do i open ports for this ip address?

frennkie, 2019-12-31:
> so how do i open ports for this ip address?

Why would you want to do that? What's your use case?

ConanWolf, 2019-12-31:
just so people can connect to my node... right? you need to port froward that port correct?

frennkie, 2019-12-31:
> just so people can connect to my node.

There is a small section covering some of this in the [FAQ](https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#why-is-my-node-address-on-the-display-red)

> **This means you can peer+openChannel with other public nodes, but other nodes cannot peer+openChannel with you.** 

The IP in your screenshot is from Comcast - so I assume that it is really the case that they are doing some kind of (carrier-grade) NAT. If that is the case then you most likely can not setup a direct NAT. 

Two ways to still get inbound connectivity on the RaspiBlitz are :

* TOR 
* SSH Reverse Tunnels

closed 2020-09-17 by rootzoll