raspiblitz/raspiblitz #949: Testnet not working due to chainsize checks

frennkie, 2020-01-02:
I just installed a testnet node and have some issues.

In `00raspiblitz.sh` the chainsize must be larger than `210000000000` - which is not the case on testnet:

```
blockchainsize=$(sudo du -shbc /mnt/hdd/${network}/ 2>/dev/null | head -n1 | awk '{print $1;}')
```

Also the copy scripts have lines which most likely only work on a mainnet node (on testnet the file will be in `/mnt/hdd/bitcoin/testnet3/blocks`):

```
count=$(sudo ls /mnt/hdd/bitcoin/blocks 2>/dev/null | grep -c '.dat')
```

frennkie, 2020-01-02:
I found another issue related to #217 

``` 
$ /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid
Error: Config setting for -rpcbind only applied on test network when in [test] section.
```

It seems that these must be set in a special [test] section for Testnet:
* rpcport=8332
* rpcbind=127.0.0.1:8332

Also for Testnet the convention would be to use 18332, **right?**

rootzoll, 2020-01-06:
This should now be fixed with merge - set to final testing.

rootzoll, 2020-01-29:
OK testnet is working so far for v1.4. Closing issue.

PR #951 (2020-01-06): change blockchain size calculation
There are some "sanity" checks that make sure that the system is in the expected state. E.g. there are checks that enough blockchain files are in the blocks directory. The assumed values only worked for mainnet.

This should fix #949.

I'm not too happy with the hardcoded values buried in several scripts.. might make sense to have a central file with "constant values" for this.
closed 2020-01-29 by rootzoll