raspiblitz/raspiblitz #950: Bitcoin.conf extra section for rpcport and rpcbind on testnet neeeded

frennkie, 2020-01-03:
I found another issue related to #217 

``` 
$ /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid
Error: Config setting for -rpcbind only applied on test network when in [test] section.
```

It seems that these must be set in a special [test] section for Testnet:
* rpcport=8332
* rpcbind=127.0.0.1:8332

Also for Testnet the convention would be to use 18332, **right?**

frennkie, 2020-01-03:
@rootzoll @openoms Using either the `[test]` or `test.xxx` syntax breaks the **sourcing** of these files:

```
$ /home/admin/config.scripts/network.txindex.sh status
/mnt/hdd/bitcoin/bitcoin.conf: line 30: main.rpcport=8332: command not found
/mnt/hdd/bitcoin/bitcoin.conf: line 31: main.rpcbind=127.0.0.1:8332: command not found
/mnt/hdd/bitcoin/bitcoin.conf: line 33: test.rpcport=18332: command not found
/mnt/hdd/bitcoin/bitcoin.conf: line 34: test.rpcbind=127.0.0.1:18332: command not found
##### STATUS TXINDEX
txindex=1
indexedToBlock=
blockchainHeight=611099
indexFinished=1
isIndexed=1
```

I found these occurrences (there might be more):
 
```
$ grep -n 'source /mnt/hdd/${network}/${network}.conf' config.scripts/* -R
config.scripts/network.txindex.sh:11:source /mnt/hdd/${network}/${network}.conf
config.scripts/network.txindex.sh:16:  source /mnt/hdd/${network}/${network}.conf
config.scripts/network.wallet.sh:11:source /mnt/hdd/${network}/${network}.conf
```

frennkie, 2020-01-03:
Fix for `config.scripts/network.txindex.sh` and `config.scripts/network.wallet.sh` in #952

frennkie, 2020-01-03:
@openoms I have not yet tested Tor on testnet for this issue. Is there anything special for `addnode` on testnet?

frennkie, 2020-01-03:
I found a thread with testnet tor nodes: https://github.com/bitcoin/bitcoin/pull/15791

frennkie, 2020-01-06:
I've tried a full circle on my test node.. -> TOR off + reboot + check; TOR on + reboot + check. Looks ok.  But I might already have made changes to some files on the `/mnt/hdd/` that will prevent from noticing some issues. Let's take this into the V1.4. RC2.

openoms, 2020-01-14:
on RC2 activating testnet after using on manniet with Tor on I get:
```
*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Tue 2020-01-14 14:46:30 GMT, end at Tue 2020-01-14 14:51:54 GMT. --
Jan 14 14:51:44 raspberrypi systemd[1]: bitcoind.service: Service RestartSec=30s expired, scheduling restart.
Jan 14 14:51:44 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 5.
Jan 14 14:51:44 raspberrypi systemd[1]: Stopped Bitcoin daemon.
Jan 14 14:51:44 raspberrypi systemd[1]: Starting Bitcoin daemon...
Jan 14 14:51:44 raspberrypi bitcoind[5139]: Error: Config setting for -addnode only applied on test network when in [test] section.
Jan 14 14:51:44 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=exited, status=1/FAILURE
Jan 14 14:51:44 raspberrypi systemd[1]: bitcoind.service: Failed with result 'exit-code'.
Jan 14 14:51:44 raspberrypi systemd[1]: Failed to start Bitcoin daemon.
1579013499 ERROR: Error found in Logs
1579013504 ERROR: Error found in Logs

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/testnet3/debug.log
tail: cannot open '/mnt/hdd/bitcoin/testnet3/debug.log' for reading: No such file or directory
```

openoms, 2020-01-14:
Needed to reinstall Tor and comment out the old addnode entries from bitcoin.conf, so it looks like this when works:

# bitcoind configuration
# mainnet/testnet
testnet=1
# Bitcoind options
server=1
daemon=1
txindex=1
disablewallet=1
# Connection settings
rpcuser=raspibolt
rpcpassword=mylonglongpassword
main.rpcport=8332
rpcallowip=127.0.0.1
main.rpcbind=127.0.0.1:8332
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333
# Raspberry Pi optimizations
dbcache=1024
maxorphantx=10
maxmempool=256
maxconnections=40
maxuploadtarget=5000
datadir=/mnt/hdd/bitcoin
#bind=127.0.0.1
#addnode=fno4aakpl6sg6y47.onion
#addnode=toguvy5upyuctudx.onion
#addnode=ndndword5lpb7eex.onion
#addnode=6m2iqgnqjxh7ulyk.onion
#addnode=5tuxetn7tar3q5kp.onion
test.rpcport=18332
test.rpcbind=127.0.0.1:18332
onlynet=onion
proxy=127.0.0.1:9050
main.bind=127.0.0.1
test.bind=127.0.0.1
dnsseed=0
dns=0
main.addnode=fno4aakpl6sg6y47.onion
main.addnode=toguvy5upyuctudx.onion
main.addnode=ndndword5lpb7eex.onion
main.addnode=6m2iqgnqjxh7ulyk.onion
main.addnode=5tuxetn7tar3q5kp.onion

openoms, 2020-01-14:
Also Blockchain Progress does not show on testnet:
[image]

It take only minutes though:
```
2020-01-14T15:17:04Z Synchronizing blockheaders, height: 1410000 (~95.09%)
2020-01-14T15:17:05Z Synchronizing blockheaders, height: 1412000 (~95.14%)
2020-01-14T15:17:06Z Synchronizing blockheaders, height: 1414000 (~95.34%)
2020-01-14T15:17:09Z Synchronizing blockheaders, height: 1416000 (~95.49%)
```

rootzoll, 2020-01-23:
Testnet looks good to me during testing. Is there anything else to close this issue?

frennkie, 2020-01-25:
I think this is finished.. I wanted to give it some more time for testing and feedback.

rootzoll, 2020-01-29:
Looks good in my tests. Closing the issue for v1.4 release.

closed 2020-01-29 by rootzoll